adafruit/circuitpython #2357: #2297 breaks status neopixel

theacodes, 2019-12-05:
It seems something about #2297 broke the status neopixel, at least for Winterbloom Sol boards.

Firmware built from previous revisions works fine, and all firmwares built after the change are affected.

The pixel works in the bootloader, but once circuitpython starts it's off. Neither the supervisor nor user code (using `neopixel_write` or `neopixel`) are able to control the neopixel.

Surprisingly, I had *one* board where the status pixel *almost* worked - it just displayed the colors incorrectly (basically wouldn't display anything below ~127 on each color channel).

deshipu, 2019-12-05:
The neopixel protocol is very timing sensitive, so the original code probably had to compensate for all those background tasks being called in the background...

theacodes, 2019-12-05:
Slight update: @jepler was able to reproduce this on SAMD builds. 

Also, here's a logic analyzer capture from the broken build when writing `(0b01010101, 0b01010101, 0b01010101)` to the pixel:

[image]

Compared to a working build:

[image]

dhalbert, 2019-12-05:
Still works on SAMD21: Trinket and Metro M0

theacodes, 2019-12-05:
Updating from the discord conversation.

@jepler has a *much* better analyzer than me. It seems the issue is that the timing is about ~10% faster:

bad (917khz):
[image]

good (800khz):
[image]

jerryneedell, 2019-12-07:
just to clarify -- this issue is not limited to the "status" neopixel -- For example, on a grand central_m4_express, I cannot operate a neopixel ring on Pin A1

jepler, 2019-12-10:
Closed via #2363

closed 2019-12-10 by jepler