adafruit/circuitpython #2359: Driver for Maxim MAX30101/MAX30102 Pulse Ox/Heart Rate Sensors

tammymakesthings, 2019-12-05:
The Maxim 30101/30102 are sensors that measure pulse ox (blood O2 saturation) and heart rate. DigiKey has an [evaluation board](https://www.digikey.com/product-detail/en/maxim-integrated/MAXREFDES117%23/MAXREFDES117%23-ND/6165562?) for this sensor, as do other vendors such as [SparkFun](https://www.sparkfun.com/products/15219) and [MakerFocus](https://www.amazon.com/gp/product/B07GGSM8SP/). As near as I can tell, all of these boards communicate using i2c, and the protocol is pretty well documented in Maxim's [data sheet](https://datasheets.maximintegrated.com/en/ds/MAX30102.pdf).

I've found an [Arduino library](https://www.arduinolibraries.info/libraries/spark-fun-max3010x-pulse-and-proximity-sensor-library) which supports this sensor, but nothing for CircuitPython. I'm planning a project which will use this sensor, so I'd like to work on porting the Arduino library to CircuitPython. (I've never written a driver for something before, so I'd also love help 😄.

makermelissa, 2019-12-05:
When I wrote my first CircuitPython library, which was a complex one, I referenced the Data Sheet and the Arduino Driver and looked at other CircuitPython libraries and the learn guides for reference.

makermelissa, 2019-12-05:
Oh and I asked questions on Discord if I got stuck, but I think that was mostly at the end with the publishing part of it.

alescoppelli, 2020-04-05:
I would like to know if anyone is still interesed to porting, from Arduino library or  maybe  make new driver in CircuitPython ( or also for MicroPython ) of  Maxim 30101/30102 chip.

tannewt, 2020-04-14:
@tammymakesthings what do you need to get started? I'm sure we can find folks to help.

BitKnitting, 2020-04-22:
@tammymakesthings I just got a maxrefdes117# from digikey.  it has the max30102.  i could help write the library if you are interested.  I so, do you have your current work within a github?  Thank you.

tammymakesthings, 2020-04-22:
> @tammymakesthings I just got a maxrefdes117# from digikey.  it has the max30102.  i could help write the library if you are interested.  I so, do you have your current work within a github?  Thank you.

I would love the help. I have a GitHub repo set up, but nothing in it yet. I will add the people who have expressed interest, and then we can figure out a path forward. Thank you

alescoppelli, 2020-04-26:
Hi to all.  I believe that it may useful have a file with raw data ( only samples data ) and to know which of them  are the heart rate and SpO2.  In this manner we can  develop  and test some algorithms before to try the sotfware in the bare-metal. Do somebody has the file or know where is possible to get it  ?

denbatch, 2020-05-26:
by using the maxim reference code, i built a *rough* circuitpython class, @tammymakesthings, which i can send to you if you're interested

younes-ammari, 2021-05-01:
> Oh and I asked questions on Discord if I got stuck, but I think that was mostly at the end with the publishing part of it.

i'm working in project that uses **max30102** using **micropython** ...... but unfortunately i didn't found its library ...... i just found the ### max30100 library and most people said it###  doesn't work !!

any help please ??

n-elia, 2021-05-06:
Hi, check my repo [here](https://github.com/n-elia/MAX30102-MicroPython-driver). Feel free to propose improvements.

younes-ammari, 2021-05-06:
Sounds great 👍
Thanks a lot

ArnavPC, 2024-01-18:
So looks like i also want to use this sensor and have found no library...

n-elia, 2024-01-18:
> So looks like i also want to use this sensor and have found no library...

https://github.com/n-elia/MAX30102-MicroPython-driver

I ported the driver. You will need to compute hr rate and spo2 on you own tho. Feel free to post your hints or solutions using the issue section.