adafruit/circuitpython #2360: nrf: PWMAudioOut: plays stereo at wrong rate

jepler, 2019-12-05:
This is also one of the problems that @urish encountered on #2337.

Reproducer: 
```import array
import audiopwmio
import audiocore
import board
import digitalio
import time

values = {
    'b': (-128, 127),
    'B': (0, 255),
    'h': (-32768, 32767),
    'H': (0, 65535),
}

def go(at = 'h', channel_count = 1):
    wave = array.array(at, [values[at][0]]) * channel_count * 24
    for i in range(len(wave)//2, len(wave)):
        wave[i] = values[at][1]
    print(wave)

    enable = digitalio.DigitalInOut(board.SPEAKER_ENABLE)
    enable.direction = digitalio.Direction.OUTPUT
    enable.value = True

    audio = audiopwmio.PWMAudioOut(board.SPEAKER)
    wave_sample = audiocore.RawSample(wave, channel_count = channel_count)

    audio.play(wave_sample, loop=True)
    time.sleep(1)
    audio.stop()
    audio.deinit()
    enable.deinit()
``` save as s8.py and import on a CPB at the REPL.  `s8.go(channel_count=1)` and `s8.go(channel_count=2) should play the same frequency, but they don't.

jepler, 2019-12-05:
A workaround for this seems to be to specify a second channel: ```audio = audiopwmio.PWMAudioOut(board.SPEAKER, right_channel=board.A1)
```

urish, 2019-12-05:
Doesn't seem to do the trick for me, here's my complete code:

```python
import board
import audiocore
import audiopwmio

print("PLAY")
mp3file = open("promo.mp3", "rb")
m = audiocore.MP3File(mp3file)
audio = audiopwmio.PWMAudioOut(board.SDA, right_channel=board.LED)
audio.play(m)
while True:
  pass
```

I can see the LED connected to the `board.LED` pin light up when the file is playing (and turn off as soon as it's done playing), so obviously some output goes there, but the sound from the speaker is still having the same issues

jepler, 2019-12-12:
Should have been closed via  #2364.

closed 2019-12-12 by jepler