adafruit/circuitpython #2370: Build information is not found within the repository

rhooper, 2019-12-11:
Neither the README.md nor the CONTRIBUTING.md have build or submodule commands in them, nor is there a shell script with them in it.

I'm probably not the only forgetful developer out there.  I like to find build instructions within the project repo in a README, COMPILING or CONTRIBUTING file.  When I run into odd issues, like submodules not updated, I find myself grepping the codebase for the appropriate git commands... to no avail.

A shell script would also be great, because global search of the project would also find "git submodule".  It would also be a great place to store common build errors, as well as test notes for future developers to see.

tannewt, 2019-12-11:
This is deliberate because the build info is in a Learn guide here: https://learn.adafruit.com/building-circuitpython/

Having it in both places would lead to instruction drift.

I'd be game to add a make command at the top level to sync (really update) submodules though. That way a git grep would find it.

rhooper, 2019-12-11:
The concern here is that not everyone thinks the same way and goes looking for a URL to get build instructions from.

Both a shell script and a small doc (say BUILDING.md) that references the build page, submodules, ports, and maybe mentions the unit tests would be wonderful.  That way there is minimal content in the repo, and some hints to get people who think differently get going or get unstuck.

tannewt, 2019-12-12:
@rhooper Please PR what you think is appropriate. Thanks!

dhalbert, 2020-01-15:
Fixed by #2387.

closed 2020-01-15 by dhalbert