adafruit/circuitpython #2374: samd audioout: paused property doesn't reflect resuming playback?

jepler, 2019-12-12:
While working on the mp3 player project, I encountered behavior that makes it look like "resume()" doesn't set "paused" back to False, at least while playing MP3s.  Investigate further and fix if necessary.

Dar-Scott, 2020-01-06:
What was you workaround?

jepler, 2020-01-06:
I manually tracked whether I had paused the audio

jepler, 2020-01-06:
Here's a reproducer that uses RawSample.  Tested on pygamer:
```
import time

import audiocore
import audioio
import board
import digitalio
import audiomp3

r = audiocore.RawSample('\x80' * 1024)
s = audioio.AudioOut(board.A0)
s.play(r, loop=True)
time.sleep(.1)
s.pause()
time.sleep(.1)
assert s.playing
assert s.paused
s.resume()
time.sleep(.1)
assert s.playing
assert not s.paused
```

Dar-Scott, 2020-01-06:
Workaround works for me on Feather M4 Express. 

Got it: not mp3 related, not new

I don't think I see it in the first play (not looping), but after a few tries in subsequent playing of different playables. I haven't explored this.

gamblor21, 2020-10-05:
I traced this around found the problem and was able to build a local copy to fix it. The actual error is in the samd-peripherals project in samd/sam_d5x_e5x/dma.c where the suspend interrupt flag was not cleared. Added `channel->CHINTFLAG.reg = DMAC_CHINTFLAG_SUSP;` to dma_resume_channel() and works fine on my ItsyBitsy M4.

I am assuming to fix it I create a fork/PR against that project since it is a submodule?

PR #3520 (2020-10-07): Update submodule to merge commit
Integrating atmel-samd peripherals submodules to fix #2374 samd audioout: paused property doesn't reflect resuming playback
closed 2020-10-07 by jepler