adafruit/circuitpython #2378: beta.1 crashes after ctrl-d after exception in REPL

dhalbert, 2019-12-12:
Same result on CPB, so it's not port-specific:

```
$ repl
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.1 on 2019-12-10; Adafruit Metro M4 Express with samd51j19
>>> [ctrl-d]
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.1 on 2019-12-10; Adafruit Metro M4 Express with samd51j19
>>> asdf
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'asdf' is not defined
>>> [ctrl-d]

FATAL: read zero bytes from port
term_exitfunc: reset failed for dev UNKNOWN: Input/output error
```

jerryneedell, 2019-12-12:
I cannot reproduce this
 on my CPB
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.1 on 2019-12-10; Adafruit Circuit Playground Bluefruit with nRF52840
>>> 
>>> 
>>> adasdsa
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'adasdsa' is not defined
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.1 on 2019-12-10; Adafruit Circuit Playground Bluefruit with nRF52840
>>> 
```

jepler, 2019-12-12:
Locally I built with gcc9 and gcc7.  I reproduce this with 9, but not with 7.

@dhalbert has already identified one specific place where code changed and made a problem (volatile is now needed to get the same result); this may be some other thing.

Perhaps we should consider reverting to the older compiler at this time.  @tannewt thoughts?

tannewt, 2019-12-12:
Has anyone reproduced this under GDB? I wouldn't rollback the compiler unless we can't figure it out.

jerryneedell, 2019-12-12:
> Locally I built with gcc9 and gcc7. I reproduce this with 9, but not with 7.
>
> @dhalbert has already identified one specific place where code changed and made a problem (volatile is now needed to get [...]

interesting -- That explains why my CPB experience is different. It was build with the old compiler.

dhalbert, 2019-12-12:
I am willing to try to figure this out. It probably reflects a real bug. Importantly, we should note these fixes in MicroPython for gcc 9 issues:

https://github.com/micropython/micropython/issues/4844
https://github.com/micropython/micropython/pull/5008

tannewt, 2019-12-12:
Ok, I reproduced it with an optimized build on nRF. The stack trace isn't helpful:

```
Breakpoint 1, 0x000292fa in reset_into_safe_mode ()
(gdb) bt
#0  0x000292fa in reset_into_safe_mode ()
#1  0x0002a3fc in softdevice_assert_handler ()
#2  0x00024a44 in ?? ()
```

I'm cherry picking the commits from https://github.com/micropython/micropython/pull/5008 now.

tannewt, 2019-12-12:
The commit from https://github.com/micropython/micropython/issues/4844 fixes this.

closed 2019-12-12 by dhalbert

kevinjwalters, 2019-12-16:
Glad to hear this is fixed, I just noticed it on `Adafruit CircuitPython 5.0.0-beta.1 on 2019-12-10; Adafruit Circuit Playground Bluefruit with nRF52840`

PR #2380 (2019-12-12): Incorporate GCC9 fixes from upstream
Thanks to @dpgeorge for the fixes upstream. Thanks to @jerryneedell for finding the issue and @dhalbert for finding the related topics upstream.

Fixes #2378