adafruit/circuitpython #2386: OnDiskBitmap: performs very poorly with rotated screens + bitmap on SD card

jepler, 2019-12-16:
On pygamer, which has a screen with rotation 270, displaying a fullscreen bitmap from SD takes about 20 seconds.

How to reproduce:
 * stick an SD card in a pygamer
 * put 001.bmp from 160.zip at /sd/001.bmp
 * run a test program to display just the bitmap

```
import busio, board, digitalio, storage, adafruit_sdcard

def mount_sd():
    spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
    cs = digitalio.DigitalInOut(board.SD_CS)
    sdcard = adafruit_sdcard.SDCard(spi, cs)
    vfs = storage.VfsFat(sdcard)
    storage.mount(vfs, "/sd")

mount_sd()

import board
import displayio
 
display = board.DISPLAY

filename = "/sd/001.bmp"
# Open the file
with open(filename, "rb") as bitmap_file:
 
    # Setup the file as the bitmap data source
    bitmap = displayio.OnDiskBitmap(bitmap_file)
 
    # Create a TileGrid to hold the bitmap
    tile_grid = displayio.TileGrid(bitmap, pixel_shader=displayio.ColorConverter())
#    tile_grid.transpose_xy = True
#    tile_grid.flip_y = True
 
    # Create a Group to hold the TileGrid
    group = displayio.Group()
 
    # Add the TileGrid to the Group
    group.append(tile_grid)
 
    # Add the Group to the Display
    display.show(group)
 
    # Loop forever so you can enjoy your image
    while True:
        pass
```

If the "transpose" property of the tilegrid is set to True, it is quite a bit faster.

Reportedly, the pyportal is much better.

Notably, pygamer has a rotation of 270, while pyportal has a rotation of 0.

hypothesis: when the bitmap is loaded OTHER THAN its on-disk order, the overhead of all the disk seeks becomes enormous, particularly when going through adafruit_sdcard.

tannewt, 2019-12-16:
I've seen this with internal flash as well. If the load isn't in on-disk order then it is super slow.

With the pixel loading code I tried to ensure it always loads from the source bitmap by row. The code to do it is here: https://github.com/adafruit/circuitpython/blob/master/shared-module/displayio/TileGrid.c#L429

bablokb, 2026-09-15:
Bringing up this old issue again. There are a few Pico-W based displays that cannot load larger bitmaps into memory. A test run with a 320x280 ST7789 connected to a Pico-W takes 75 (!) seconds to update the image (loaded from a SD-card).

On disk order is certainly one issue: bitmaps are stored bottom-up, TileGrid processes top-down, so rotation matters. Independent of this, the code currently runs one lseek+read *per pixel*, even if two pixels share a byte (e.g. for ACeP displays).

Caching a row within OnDiskBitmap would reduce the overhead dramatically. My example would only need 320 lseek+reads instead of 89.600. The cost would be 2x320 = 640 bytes for the buffer. For a 800x480 ACeP display (e.g. Pimoroni Inky-Frame 7.3") the buffer would only be 400 bytes.

closed 2026-09-25 by tannewt