adafruit/circuitpython #2392: Stack depth not sufficient for SAMD21 - specifically CPX with frozen CPX lib

kattni, 2019-12-16:
Encountered this when updating the Circuit Playground library to be able to detect board type. The stack size on the CPX is no longer large enough to handle the restructured Circuit Playground library. @dhalbert tested and found increasing the stack size to 4504 from 4096 resolved the issue. @tannewt agreed increasing the stack size was acceptable.

This needs to be done on both 5.x and 4.x as we are updating the Circuit Playground examples to use the new import mechanism. Updating 4.x should wait until the Circuit Playground library example updates are complete. 5.x can happen at any time as the library is not frozen into Bluefruit, and we're in a frequent update cycle at the moment.

ladyada, 2019-12-20:
i think this was resolved?

dhalbert, 2019-12-20:
Fixed by #2407.

dhalbert, 2019-12-20:
Fxed in 4.x by #2407. awaiting 5.x fix.

dhalbert, 2019-12-20:
Fixed by #2407 and #2393.

closed 2019-12-20 by dhalbert