adafruit/circuitpython #2412: robohatmm1 in 4.1.x branch needs to be renamed to robohatmm1_m4 to match master

makermelissa, 2019-12-19:
This is causing an issue on the build names and how the boards show up on circuitpython.org

makermelissa, 2019-12-19:
Unfortunately, I'm not the greatest at dealing with multiple branches on such a complex repo, so I'm not sure the best way to go about it.

dhalbert, 2019-12-19:
So after it's fixed manually by https://github.com/adafruit/circuitpython-org/pull/344, we need to rename the board before doing any more 4.x releases (per discussion with @makermelissa in discord).

makermelissa, 2019-12-19:
It might need to be fixed before the next 5.x release if the data files are completely regenerated.

dhalbert, 2019-12-19:
I can add renamed assets in the 4.1.2 release to match the new board name, so that the links will continue to work. Where are the board names extracted from: the last release to generate the circuitpyhon-org PR, or somewhere else?

makermelissa, 2019-12-19:
You don't need to rename the assets. I'm pointing to the current ones.

makermelissa, 2019-12-19:
I think you just need to change the folder name under ports/atmel-samd/boards

dhalbert, 2019-12-20:
@wallarug See above. The robohatmm1 in the 4.x tree is a SAMD21 board. Is that discontinued? Are you expecting everyone to use CircuitPython 5.x now? If that's true, we could just remove the board from 4.x. Thanks for looking.

wallarug, 2019-12-20:
Hey @makermelissa  and @dhalbert 

Please remove robothatmm1 from the 4.x tree.   We are only making the SAMD51.

If you feel so inclined, the robohatmm1_m4 from the master branch can be put onto 4.x as well.

wallarug, 2019-12-22:
Can this issue be closed now?  @dhalbert @makermelissa

makermelissa, 2019-12-22:
Yeah, I'm good with closing it.

closed 2019-12-22 by makermelissa

dhalbert, 2019-12-22:
Fixed by #2415.

PR #2415 (2019-12-20): replace robohatmm1 with robohatmm1_m4
Fixes #2412; M4 board added by request of @wallarug (@wallarug Please review.)

@wallarug I'm not sure when we will do another 4.x release, but there may be one due to some library changes.