adafruit/circuitpython #2417: Lockup with displayio + other activity on SD card

jepler, 2019-12-20:
In jeplayer, I ran into the following hang when I had a bitmap on SD and I also manually called refresh.  The problem appears to be linked to the fact that the same file is being "f_read" recursively; these reads both make calls out into Python to the single, non-reentrant SD card library.  I suspect, but don't know how to verify, that the inner call is waiting for the SPI device to become available.

Should it be the case that display background is inhibited during an explicit refresh?

```
#0  mp_map_lookup (map=map@entry=0x20004ee4, index=index@entry=0x1bc6, lookup_kind=MP_MAP_LOOKUP) at ../../py/map.c:151
#1  0x0002c294 in mp_obj_instance_load_attr (dest=0x2002ea4c, attr=888, self_in=<optimized out>) at ../../py/objtype.c:599
#2  mp_obj_instance_attr (self_in=0x20004ee0, attr=888, dest=0x2002ea4c) at ../../py/objtype.c:820
#3  0x00027fc0 in mp_load_method (base=0x20004ee0, attr=888, dest=0x2002ea4c) at ../../py/runtime.c:1091
#4  0x00039b8c in mp_load_attr (attr=<optimized out>, base=<optimized out>) at ../../py/runtime.c:929
#5  mp_execute_bytecode (code_state=code_state@entry=0x2002ead0, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:331
#6  0x0003b518 in fun_bc_call (self_in=0x20028570, n_args=<optimized out>, n_kw=128, args=0x200161c0) at ../../py/objfun.c:284
#7  0x0003aa82 in mp_call_method_n_kw (args=<optimized out>, n_kw=0, n_args=0) at ../../py/runtime.c:637
#8  mp_execute_bytecode (code_state=code_state@entry=0x200161a0, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:584
#9  0x0003b518 in fun_bc_call (self_in=0x20027480, n_args=<optimized out>, n_kw=537062740, args=0x20004fcc) at ../../py/objfun.c:284
#10 0x00018df4 in disk_read (pdrv=<optimized out>, buff=<optimized out>, sector=<optimized out>, count=<optimized out>, count=<optimized out>, 
    sector=<optimized out>, buff=<optimized out>, pdrv=<optimized out>) at ../../extmod/vfs_fat_diskio.c:83
#11 0x00018e3a in move_window (fs=<optimized out>, sector=<optimized out>, sector=<optimized out>, fs=<optimized out>) at ../../lib/oofatfs/ff.c:888
#12 0x0001b272 in f_read (fp=0x200113c4, buff=buff@entry=0x2002ed58, btr=btr@entry=3, br=br@entry=0x2002ed54) at ../../lib/oofatfs/ff.c:3424
#13 0x0001c048 in common_hal_displayio_ondiskbitmap_get_pixel (y=<optimized out>, x=1, self=0x20011740) at ../../shared-module/displayio/OnDiskBitmap.c:138
#14 displayio_tilegrid_fill_area (buffer=<optimized out>, mask=<optimized out>, area=<optimized out>, colorspace=<optimized out>, self=0x20011880)
    at ../../shared-module/displayio/TileGrid.c:436
#15 displayio_group_fill_area (self=0x20011080, colorspace=colorspace@entry=0x20001762 <displays+162>, area=0x2002f014, area@entry=0x2002f054, mask=0x2002edb8, 
    mask@entry=0x200016c0 <displays>, buffer=0x2002ede0, buffer@entry=0x13) at ../../shared-module/displayio/Group.c:321
#16 0x0001c432 in displayio_display_core_fill_area (buffer=0x13, mask=0x200016c0 <displays>, area=0x2002f054, self=0x20001718 <displays+88>)
    at ../../shared-module/displayio/display_core.c:316
#17 _refresh_area (area=0x2000174c <displays+140>, self=0x20001710 <displays+80>) at ../../shared-module/displayio/Display.c:279
#18 _refresh_display (self=self@entry=0x20001710 <displays+80>) at ../../shared-module/displayio/Display.c:305
#19 0x0001de78 in displayio_display_background (self=0x20001710 <displays+80>) at ../../shared-module/displayio/Display.c:385
#20 displayio_background () at ../../shared-module/displayio/__init__.c:46
#21 run_background_tasks () at background.c:84
#22 supervisor_run_background_tasks_if_tick () at ../../supervisor/shared/tick.c:88
#23 0x0003ac56 in mp_execute_bytecode (code_state=code_state@entry=0x20016120, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:1295
#24 0x0003b518 in fun_bc_call (self_in=0x20027ee0, n_args=<optimized out>, n_kw=1216, args=0x200160b8) at ../../py/objfun.c:284
#25 0x0003a484 in mp_call_method_n_kw (args=<optimized out>, n_kw=<optimized out>, n_args=<optimized out>) at ../../py/runtime.c:637
#26 mp_execute_bytecode (code_state=code_state@entry=0x200160a0, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:1015
#27 0x0003b518 in fun_bc_call (self_in=0x200276e0, n_args=<optimized out>, n_kw=1217, args=0x20011d50) at ../../py/objfun.c:284
#28 0x0003a484 in mp_call_method_n_kw (args=<optimized out>, n_kw=<optimized out>, n_args=<optimized out>) at ../../py/runtime.c:637
#29 mp_execute_bytecode (code_state=code_state@entry=0x20011d30, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:1015
#30 0x0003b518 in fun_bc_call (self_in=0x20027480, n_args=<optimized out>, n_kw=537064532, args=0x20004fcc) at ../../py/objfun.c:284
#31 0x00018df4 in disk_read (pdrv=<optimized out>, buff=<optimized out>, sector=<optimized out>, count=<optimized out>, count=<optimized out>, 
    sector=<optimized out>, buff=<optimized out>, pdrv=<optimized out>) at ../../extmod/vfs_fat_diskio.c:83
#32 0x00018e3a in move_window (fs=<optimized out>, sector=<optimized out>, sector=<optimized out>, fs=<optimized out>) at ../../lib/oofatfs/ff.c:888
#33 0x0001b272 in f_read (fp=0x200113c4, buff=buff@entry=0x2002f458, btr=btr@entry=3, br=br@entry=0x2002f454) at ../../lib/oofatfs/ff.c:3424
#34 0x0001c048 in common_hal_displayio_ondiskbitmap_get_pixel (y=<optimized out>, x=34, self=0x20011740) at ../../shared-module/displayio/OnDiskBitmap.c:138
#35 displayio_tilegrid_fill_area (buffer=<optimized out>, mask=<optimized out>, area=<optimized out>, colorspace=<optimized out>, self=0x20011880)
    at ../../shared-module/displayio/TileGrid.c:436
#36 displayio_group_fill_area (self=0x20011080, colorspace=colorspace@entry=0x20001762 <displays+162>, area=0x2002f714, area@entry=0x2002f754, mask=0x2002f4b8, 
    mask@entry=0x20001710 <displays+80>, buffer=0x2002f4e0, buffer@entry=0x20001710 <displays+80>) at ../../shared-module/displayio/Group.c:321
#37 0x0001c432 in displayio_display_core_fill_area (buffer=0x20001710 <displays+80>, mask=0x20001710 <displays+80>, area=0x2002f754, self=0x20001718 <displays+88>)
    at ../../shared-module/displayio/display_core.c:316
#38 _refresh_area (area=0x2000174c <displays+140>, self=0x20001710 <displays+80>) at ../../shared-module/displayio/Display.c:279
#39 _refresh_display (self=self@entry=0x20001710 <displays+80>) at ../../shared-module/displayio/Display.c:305
#40 0x0002a4fa in common_hal_displayio_display_refresh (maximum_ms_per_real_frame=4294967295, target_ms_per_frame=<optimized out>, self=0x20001710 <displays+80>)
    at ../../shared-module/displayio/Display.c:353
#41 displayio_display_obj_refresh (n_args=<optimized out>, pos_args=<optimized out>, kw_args=<optimized out>) at ../../shared-bindings/displayio/Display.c:252
#42 0x00019e74 in fun_builtin_var_call (self_in=<optimized out>, n_args=<optimized out>, n_kw=<optimized out>, args=<optimized out>, args=<optimized out>, 
    n_kw=<optimized out>, n_args=<optimized out>, self_in=<optimized out>) at ../../py/objfun.c:122
#43 0x0003a484 in mp_call_method_n_kw (args=<optimized out>, n_kw=<optimized out>, n_args=<optimized out>) at ../../py/runtime.c:637
#44 mp_execute_bytecode (code_state=code_state@entry=0x20010fb0, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:1015
#45 0x0003b518 in fun_bc_call (self_in=0x200048b0, n_args=<optimized out>, n_kw=242825, args=0x20010de8) at ../../py/objfun.c:284
#46 0x0003a50a in mp_call_function_n_kw (args=0x20010de8, n_kw=<optimized out>, n_args=<optimized out>, fun_in=0x200048b0) at ../../py/runtime.c:622
#47 mp_execute_bytecode (code_state=code_state@entry=0x20010dc0, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:932
#48 0x0003b518 in fun_bc_call (self_in=0x20004900, n_args=<optimized out>, n_kw=242825, args=0x2002fa50) at ../../py/objfun.c:284
#49 0x0003a50a in mp_call_function_n_kw (args=0x2002fa50, n_kw=<optimized out>, n_args=<optimized out>, fun_in=0x20004900) at ../../py/runtime.c:622
#50 mp_execute_bytecode (code_state=code_state@entry=0x2002fa38, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:932
#51 0x0003b518 in fun_bc_call (self_in=0x20004920, n_args=<optimized out>, n_kw=242825, args=0x200042c8) at ../../py/objfun.c:284
#52 0x0003a50a in mp_call_function_n_kw (args=0x200042c8, n_kw=<optimized out>, n_args=<optimized out>, fun_in=0x20004920) at ../../py/runtime.c:622
[... cut ...]
#61 0x00037b44 in parse_compile_execute (source=source@entry=0x48aa7, input_kind=input_kind@entry=MP_PARSE_FILE_INPUT, exec_flags=exec_flags@entry=32, 
    result=result@entry=0x2002ffb8) at ../../lib/utils/pyexec.c:114
#62 0x00037c68 in pyexec_file (result=0x2002ffb8, filename=0x48aa7 "main.py") at ../../lib/utils/pyexec.c:529
#63 maybe_run_list (filenames=filenames@entry=0x200003c8 <supported_filenames>, exec_result=exec_result@entry=0x2002ffb8) at ../../main.c:183
#64 0x000385ce in run_code_py (safe_mode=<optimized out>) at ../../main.c:241
#65 main () at ../../main.c:462
```

dhalbert, 2019-12-20:
> Should it be the case that display background is inhibited during an explicit refresh?

That would certainly make sense, since both call `_refresh_display()`. There's a test for recursivity in `displayio_background()` and there should probably be one in `_refresh_display()` as well. Maybe they could share the recursive-check flag. Or `displayio_background()` would depend on a new recursivity check in `_refresh_display()`.

jepler, 2019-12-20:
Fixing displayio to not recurse itself is one thing, but it's also not clear how a developer of pure python code can avoid a problem!    For instance, I also observed a crash where the stack indicates Python code was doing an `os.stat` of the SD card, which entered background tasks, and displayio tried to read a bitmap also from SD.

dhalbert, 2019-12-20:
Bleh. So SD card operations really need to be serialized.

jepler, 2019-12-20:
Right now it feels like a tough problem and I'm adding workarounds in JEplayer.

jepler, 2019-12-20:
Another scenario is displayio refresh + mp3:
```
#0  mp_load_method_maybe (obj=0x20007080, attr=888, dest=0x2002ed2c) at ../../py/runtime.c:1063
#1  0x00027fc0 in mp_load_method (base=0x20007080, attr=888, dest=0x2002ed2c) at ../../py/runtime.c:1091
#2  0x00039b8c in mp_load_attr (attr=<optimized out>, base=<optimized out>) at ../../py/runtime.c:929
#3  mp_execute_bytecode (code_state=code_state@entry=0x2002edb0, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:331
#4  0x0003b518 in fun_bc_call (self_in=0x200281a0, n_args=<optimized out>, n_kw=128, args=0x20011990) at ../../py/objfun.c:284
#5  0x0003aa82 in mp_call_method_n_kw (args=<optimized out>, n_kw=0, n_args=0) at ../../py/runtime.c:637
#6  mp_execute_bytecode (code_state=code_state@entry=0x20011970, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:584
#7  0x0003b518 in fun_bc_call (self_in=0x200270b0, n_args=<optimized out>, n_kw=537063372, args=0x2000731c) at ../../py/objfun.c:284
#8  0x00018df4 in disk_read (pdrv=<optimized out>, buff=<optimized out>, sector=<optimized out>, count=<optimized out>, count=<optimized out>, 
    sector=<optimized out>, buff=<optimized out>, pdrv=<optimized out>) at ../../extmod/vfs_fat_diskio.c:83
#9  0x00018e3a in move_window (fs=<optimized out>, sector=<optimized out>, sector=<optimized out>, fs=<optimized out>) at ../../lib/oofatfs/ff.c:888
#10 0x0001b272 in f_read (fp=0x20007d04, buff=buff@entry=0x200095cc, btr=btr@entry=1044, br=br@entry=0x2002efcc) at ../../lib/oofatfs/ff.c:3424
#11 0x0001b41a in mp3file_update_inbuf (self=self@entry=0x200083a0) at ../../shared-module/audiomp3/MP3File.c:67
#12 0x0001ba56 in mp3file_skip_id3v2 (self=self@entry=0x200083a0) at ../../shared-module/audiomp3/MP3File.c:94
#13 0x0001bb7a in audiomp3_mp3file_get_buffer (self=0x200083a0, single_channel=<optimized out>, channel=<optimized out>, bufptr=<optimized out>, 
    buffer_length=0x2002f020) at ../../shared-module/audiomp3/MP3File.c:298
#14 0x0001dc48 in audio_dma_load_next_block (dma=0x2000de8c) at audio_dma.c:122
#15 0x0001dd6e in audio_dma_background () at audio_dma.c:361
#16 run_background_tasks () at background.c:81
#17 supervisor_run_background_tasks_if_tick () at ../../supervisor/shared/tick.c:88
#18 0x0003ac56 in mp_execute_bytecode (code_state=code_state@entry=0x200118f0, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:1295
#19 0x0003b518 in fun_bc_call (self_in=0x20027b10, n_args=<optimized out>, n_kw=1230, args=0x20011888) at ../../py/objfun.c:284
#20 0x0003a484 in mp_call_method_n_kw (args=<optimized out>, n_kw=<optimized out>, n_args=<optimized out>) at ../../py/runtime.c:637
#21 mp_execute_bytecode (code_state=code_state@entry=0x20011870, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:1015
#22 0x0003b518 in fun_bc_call (self_in=0x20027310, n_args=<optimized out>, n_kw=1231, args=0x200109d0) at ../../py/objfun.c:284
#23 0x0003a484 in mp_call_method_n_kw (args=<optimized out>, n_kw=<optimized out>, n_args=<optimized out>) at ../../py/runtime.c:637
#24 mp_execute_bytecode (code_state=code_state@entry=0x200109b0, inject_exc=<optimized out>, inject_exc@entry=0x0) at ../../py/vm.c:1015
#25 0x0003b518 in fun_bc_call (self_in=0x200270b0, n_args=<optimized out>, n_kw=537064532, args=0x2000731c) at ../../py/objfun.c:284
#26 0x00018df4 in disk_read (pdrv=<optimized out>, buff=<optimized out>, sector=<optimized out>, count=<optimized out>, count=<optimized out>, 
    sector=<optimized out>, buff=<optimized out>, pdrv=<optimized out>) at ../../extmod/vfs_fat_diskio.c:83
#27 0x00018e3a in move_window (fs=<optimized out>, sector=<optimized out>, sector=<optimized out>, fs=<optimized out>) at ../../lib/oofatfs/ff.c:888
#28 0x0001b272 in f_read (fp=0x2000e6e4, buff=buff@entry=0x2002f458, btr=btr@entry=3, br=br@entry=0x2002f454) at ../../lib/oofatfs/ff.c:3424
#29 0x0001c048 in common_hal_displayio_ondiskbitmap_get_pixel (y=<optimized out>, x=114, self=0x2000eab0) at ../../shared-module/displayio/OnDiskBitmap.c:138
#30 displayio_tilegrid_fill_area (buffer=<optimized out>, mask=<optimized out>, area=<optimized out>, colorspace=<optimized out>, self=0x2000ebf0)
    at ../../shared-module/displayio/TileGrid.c:436
#31 displayio_group_fill_area (self=0x2000e090, colorspace=colorspace@entry=0x20001762 <displays+162>, area=0x2002f714, area@entry=0x2002f754, mask=0x2002f4b8, 
    mask@entry=0x20001710 <displays+80>, buffer=0x2002f4e0, buffer@entry=0x20001710 <displays+80>) at ../../shared-module/displayio/Group.c:321
#32 0x0001c432 in displayio_display_core_fill_area (buffer=0x20001710 <displays+80>, mask=0x20001710 <displays+80>, area=0x2002f754, self=0x20001718 <displays+88>)
    at ../../shared-module/displayio/display_core.c:316
#33 _refresh_area (area=0x2000174c <displays+140>, self=0x20001710 <displays+80>) at ../../shared-module/displayio/Display.c:279
#34 _refresh_display (self=self@entry=0x20001710 <displays+80>) at ../../shared-module/displayio/Display.c:305
#35 0x0002a4fa in common_hal_displayio_display_refresh (maximum_ms_per_real_frame=4294967295, target_ms_per_frame=<optimized out>, self=0x20001710 <displays+80>)
    at ../../shared-module/displayio/Display.c:353
#36 displayio_display_obj_refresh (n_args=<optimized out>, pos_args=<optimized out>, kw_args=<optimized out>) at ../../shared-bindings/displayio/Display.c:252
#37 0x00019e74 in fun_builtin_var_call (self_in=<optimized out>, n_args=<optimized out>, n_kw=<optimized out>, args=<optimized out>, args=<optimized out>, 
```

jepler, 2020-02-03:
We'll put a release note that "here be dragons" and not block 5.0.0 with this.

dhalbert, 2020-02-03:
@jepler Could you write a few sentences about this for the draft release: how it might come about and how to avoid it? Thanks! https://github.com/adafruit/circuitpython/releases/tag/untagged-b5510d684adf6cd75f14 (if that link is broken try just the release link).

jepler, 2020-02-28:
While I didn't end up prototyping it, I suspect that my idea of blocking the audio background task when an sd card I/O is going on will lead to audio stutters, as it's very easy to cause single I/Os that take >>20ms.  Since we can't tell whether the background task is also going to need the same SPI bus (this is what causes the deadlock), we'll be making all audio background suffer for a tiny sliver of all cases, and with no way to make it better.  Thus, I  think leaving up the "here be dragons" sign until we fix it with SD I/O written in C that can cooperate with background tasks is the better alternative.

tannewt, 2020-07-13:
@jepler Does your SDIO work improve this?

jepler, 2020-07-14:
Yes, sdcard-in-core and sdio both improve on this, unless I overlooked something.  I did not specifically test for this, the following's based on my thoughts about how shifting things into the C core will relate to background tasks.

If you use either of these instead of adafruit_sdcard, then it is not possible to enter background tasks during an SD card I/O operation, so the case of reading/writing SD in the foreground will not cause any problem (other than a delay) if background tasks also cause access.

If you use other SPI devices such as adafruit_busdevice in the foreground, lockups could still occur.

Still, the work that has been done for 6.0.0 markedly improves things and the original steps to reproducing no longer apply.

tannewt, 2020-07-14:
So, should we close this or leave it open with a new way to reproduce?

jepler, 2020-07-16:
The bug, as described here, has been fixed.  There are other ways with multiple devices on the SPI bus that can cause similar problems, but these scenarios are much less likely to occur, since the two most common devices (displays and SD cards) can now be used without deadlocks.

The remaining scenario is when the same SPI bus is used by foreground Python code that locks the SPI bus (including using bus_device) and is also used by display or by sdcardio.

closed 2020-07-16 by jepler