adafruit/circuitpython #2421: STM32F405 doesn't mount as CIRCUITPY on macOS with 5.0.0-beta.2

anecdata, 2019-12-21:
Using macOS Catalina (10.15.2), can't get the STM32F405 to mount as CIRCUITPY under 5.0.0-beta.2. Using 5.0.0-beta.1, it mounts fine.

A one-liner code.py with just a single print statement seems to run to completion (pulsing green NeoPixel), but no sign of the CIRCUITPY device using `system_profiler SPUSBDataType`, or as a USB Volume. In bootloader mode, it shows up fine (via `system_profiler SPUSBDataType` or `dfu-util`). Tried re-downloading beta.2, reloading into device, power-cycling, etc.

Switching back and forth between beta.1 and beta.2, the behavior looks consistent: with beta.1 it always mounts, with beta.2 it never mounts.

jerryneedell, 2019-12-21:
I have reproduced this on a Linux system -- no CIRCUITPY or /dev/ttyACM0 device
The issue actually began before the 5.0beta-2 tag  -- the first build I have that fails to mount is:
adafruit-circuitpython-feather_stm32f405_express-en_US-20191218-25df6e6.bin

Interestingly -- there is no information in dmesg when I plug in or reset the board
The status neopixel is "breathing" green -- looks like a normal startup but no /dev/ttyACM0

jerryneedell, 2019-12-21:
It appear that PR #2395 introduced this issue
 adafruit-circuitpython-feather_stm32f405_express-en_US-20191218-25df6e6.bin
does not mount -- no information via dmesg

adafruit-circuitpython-feather_stm32f405_express-en_US-20191218-f2b1aee.bin
does mount normally

dhalbert, 2019-12-21:
Tagging @hathach here: could you take a look. There were many TinyUSB changes in #2395 as well as the nrfx update.

phrogger, 2019-12-22:
I have reproduced this on a Windows 10 System -- no CIRCUITPY

hathach, 2019-12-23:
This will be fixed by #2379 , tinyusb is now leave the VBUS sense configuration up to application to decide e.g STM32f405 requires to set hw vbus sense, while stm32f411 blackpill insist to turn it off.

jerryneedell, 2019-12-23:
tested on both Linux and MacOS Catalina -- works for me!

phrogger, 2019-12-23:
20191223-21cbd5c.bin restored operation on Windows 10.

hierophect, 2019-12-23:
Sorry for being a little late to the party here. Seems like it's just the same VBUS issue the #2379 PR was designed to fix, that just got ahead of our updates for a minute. I've gotten my text fixes in with #2424 and it seems like it's working for everybody again so I think this can be closed.

closed 2019-12-23 by hierophect

hathach, 2019-12-24:
@hierophect no problems at all, It is Christmas time :)