adafruit/circuitpython #2422: STM32F405 under CP 5.0.0 Beta1 SPI Initialization Problem

phrogger, 2019-12-22:
CP5.0.0 Beta1 running on STM32F405 will not initialize the SPI bus using busio.SPI(), but will run using board.SPI().  

Bootout: Adafruit CircuitPython 5.0.0-beta.1 on 2019-12-10; Feather STM32F405 Express with STM32F405RG

    import board
    import busio
    import digitalio

    spi = busio.SPI(board.SCK, MOSI=board.MOSI)    # setup SPI bus

    # Results in:

    main.py output:
    Traceback (most recent call last):
        File "main.py", line 10, in <module>
    ValueError: Invalid SPI pin selection`

Eliminating the MOSI declaration does not change the result, so it appears that the declaration of SCK is causing the exception.

Whereas, using spi = board.SPI() seems to work.

ladyada, 2019-12-22:
probably dup of https://github.com/adafruit/circuitpython/issues/2309

phrogger, 2019-12-22:
Yes, I missed that #2309 had already been reported. Likely the same problem.

closed 2019-12-22 by ladyada