adafruit/circuitpython #2425: SAMD21 Xplained Pro support

riegaz, 2019-12-23:
Is this Atmel Evaluation Kit supported? I was not able to get it going.

Any help is welcome.

ladyada, 2019-12-23:
not yet but it ought to work with internal storage (unless it has a SPI flash on it?)

riegaz, 2019-12-24:
I didn't get it working flashing any bootloader onto it. It has a serial flash installed but I would prefer it working qith the internal storage.

ladyada, 2019-12-25:
ok start with the UF2 bootloader, you can probably look at the Trinket M0 or Feather M0 Basic for inspiration :)

tannewt, 2020-01-02:
Check out this guide about adding new boards to CircuitPython: https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython

barawn, 2020-03-04:
I've gotten this working (although I just grabbed USB VID/PIDs, so don't ask for the binary :) ). The UF2 portion is painless - just copy over a board_config.h from the Feather M0, change LED_PIN to PB_30 (and keep in mind it's inverted, so it'll be opposite whatever the Feather's does), change CHIP_VARIANT to SAMD21J18A, and build it.

The CircuitPython part requires fixes in the asf4 repository (this has an open pull request to fix it, should be merged soon I imagine) and the samd-peripherals repository (already fixed in master, not updated in circuitpython master).

Once  that's done creating a board with CHIP_VARIANT = SAMD21J18A builds fine. The SPI flash is an AT25DF081A, on PB22/PB16/PB23/PA13 (MOSI/MISO/SCK/CS), so just update those in mpconfigboard.mk/mpconfigboard.h.

glucee, 2020-05-22:
@barawn can you share a bin file for that.

tannewt, 2020-05-26:
@barawn Feel free to mention me on anything pending that still needs to be reviewed. I can provide a USB VID and PID too. So, feel free to make a PR and we can sort things out there.

dhalbert, 2025-01-31:
Closing due to no progress. If anyone wants to do a PR that would be fine.

closed 2025-01-31 by dhalbert