adafruit/circuitpython #2426: nrf: hang in nrfx_spim_xfer

jepler, 2019-12-25:
I ran the following code:
```
import board
import busio
import digitalio
import time

b = busio.SPI(board.SCK, MOSI=board.MOSI)
stb = digitalio.DigitalInOut(board.D5)

stb.switch_to_output(True)
d = bytearray(4)
d[0] = d[1] = d[2] = d[3] = 0xff
b.try_lock()
b.write(d)
```
and it hangs during `write`.  The last few frames of the traceback are:
```
Program received signal SIGTRAP, Trace/breakpoint trap.
nrfx_spim_xfer (p_instance=0x200044a0 <spim_peripherals>, p_xfer_desc=0x2003fd28, flags=0)
    at nrfx/drivers/src/nrfx_spim.c:581
581            while (!nrf_spim_event_check(p_spim, NRF_SPIM_EVENT_END)){}
(gdb) where
#0  nrfx_spim_xfer (p_instance=0x200044a0 <spim_peripherals>, p_xfer_desc=0x2003fd28, flags=0)
    at nrfx/drivers/src/nrfx_spim.c:581
#1  0x00055d6e in common_hal_busio_spi_write (self=self@entry=0x20008680, data=0x200086d0 "\377\377\377\377", 
    len=<optimized out>) at common-hal/busio/SPI.c:249
```
it's not clear what is stopping the SPI transaction from finishing.  Maybe out-only SPI is just broken?  Surely not all SPI is this broken?

dhalbert, 2019-12-25:
Which board and version is this, and have you tried an earlier version? It did used to work :). This may be due to upgrading nrfx.

dhalbert, 2019-12-25:
I looked over the nrfx 2.0 changes and the changes made to CircuitPython for the update, and I don't see anything obvious, so more debugging is in order.

jepler, 2019-12-25:
Turns out it was my wiring.  I was using an experimental board and it's simply not right; in effect, I had hooked SCK to GND.  For better or for worse, this makes nrfx_spim_xfer hang forever(!)  I initially dismissed the wiring a possible cause, because "obviously" the functioning of the code couldn't depend on the input value on SCK or MOSI.

closed 2019-12-25 by jepler

dhalbert, 2019-12-25:
!! Yes, I just tried this, and it worked for me on an unconnected Feather '840. That's pretty odd.