adafruit/circuitpython #2429: pybadge_airlift kills pybadge, requires reversion to non-airlift firmware

hukuzatuna, 2019-12-25:
Building the ports/atmel-samd/pybadge_airlift firmware (as of the morning of 12/25/2019) and then loading it onto a PyBadge (I don't see a hardware rev designation) causes the board to stop working. By that, I mean CIRCUITPY doesn't reappear as a mass storage device the display doesn't initialize (it stays dark with no backlight). The bootloader, however, works, so I can load the ports/atmel-samd/pybadge firmware (built at the same time) to restore functionality.

Let me know if you need any additional information or debugging assistance.

ladyada, 2019-12-25:
its a totally different board, so its not expected to work. what were you expecting to happen?

hukuzatuna, 2019-12-25:
Ahhhh, ok. I put an AirLift FeatherWing on it and it timed out trying to find the FeatherWing. I thought it was SPI contention, since I thought the display and the Airlift were both SPI devices, and so I figured the pybadge_airlift was the supported configuration. I'll go back to trying the PyBadge + AirLift FeatherWing combination.

ladyada, 2019-12-25:
ok!

closed 2019-12-25 by ladyada

hukuzatuna, 2019-12-25:
@ladyada Oh for heaven's sake. I just tried the esp32_simpletest.py code and it works fine. I'm not sure where I went wrong before, but it seems good.