adafruit/circuitpython #2433: STM32: possible problem with strings and adafruit_minimqtt library (AdafruitIO)

hukuzatuna, 2019-12-28:
I have the Feather STM32F405 Express board. I built firmware for it on Christmas. I have an LPS3x bord connected via QT. If I just print data, everything is fine. If I try to push the data to AdafruitIO, though, I get an error "AttributeError: 'str' object has no attribute 'encode'. When I run 
 adafruit/Adafruit_CircuitPython_AdafruitIO/examples/mqtt/adafruit_io_simpletest.py it throws the same error and backtrace. Here's the backtrace, code to follow down below.
```
Press any key to enter the REPL. Use CTRL-D to reload.soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
Traceback (most recent call last):
  File "code.py", line 93, in <module>
  File "adafruit_minimqtt.py", line 121, in __init__
AttributeError: 'str' object has no attribute 'encode'
```
Then I thought maybe something had changed, so I went back to the Adafruit IO guides and could not find resolution.

Then I went to the library bundle 5.x source as of 20191223 and looked at line 121 in adafruit_minimqtt.py. (If you need me to include a snippet here, I will.) There didn't seem t be anything unusual, but I did see the .encode() method that was causing the problem.

Then I just connected to the REPL, created a string and called the same encode method:
```
Adafruit CircuitPython 5.0.0-beta.2-13-g7387f6092 on 2019-12-27; Feather STM32F405 Express with STM32F405RG
>>> foo = "heresastringtotestwith"
>>> foo.encode('utf-u')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'str' object has no attribute 'encode'
>>> 
```
The last couple of issues I raised LadyAda had to exercise extreme patience with my bumbling, so I hope this is more cohesive and helpful. I still assume I'm just doing something stupid, but people on Discord don't seem to think so.

Here is the code that generated the original backtrace:
```
# Example of using the Adafruit IO CircuitPython MQTT client
# to subscribe to an Adafruit IO feed and publish random data
# to be received by the feed.
#
# Example by Tony DiCola for Adafruit Industries
# Modified by Brent Rubell for Adafruit Industries, 2019
import time
from random import randint

import board
import busio
from adafruit_esp32spi import adafruit_esp32spi
from adafruit_esp32spi import adafruit_esp32spi_wifimanager
import adafruit_esp32spi.adafruit_esp32spi_socket as socket
from digitalio import DigitalInOut
import neopixel
from adafruit_io.adafruit_io import IO_MQTT
from adafruit_minimqtt import MQTT

### WiFi ###

# Get wifi details and more from a secrets.py file
try:
    from secrets import secrets
except ImportError:
    print("WiFi secrets are kept in secrets.py, please add them there!")
    raise

# If you have an externally connected ESP32:
esp32_cs = DigitalInOut(board.D13)
esp32_ready = DigitalInOut(board.D11)
esp32_reset = DigitalInOut(board.D12)

spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
esp = adafruit_esp32spi.ESP_SPIcontrol(spi, esp32_cs, esp32_ready, esp32_reset)
"""Use below for Most Boards"""
status_light = neopixel.NeoPixel(
    board.NEOPIXEL, 1, brightness=0.2
)  # Uncomment for Most Boards
"""Uncomment below for ItsyBitsy M4"""
# status_light = dotstar.DotStar(board.APA102_SCK, board.APA102_MOSI, 1, brightness=0.2)
# Uncomment below for an externally defined RGB LED
# import adafruit_rgbled
# from adafruit_esp32spi import PWMOut
# RED_LED = PWMOut.PWMOut(esp, 26)
# GREEN_LED = PWMOut.PWMOut(esp, 27)
# BLUE_LED = PWMOut.PWMOut(esp, 25)
# status_light = adafruit_rgbled.RGBLED(RED_LED, BLUE_LED, GREEN_LED)
wifi = adafruit_esp32spi_wifimanager.ESPSPI_WiFiManager(esp, secrets, status_light)

# Define callback functions which will be called when certain events happen.
# pylint: disable=unused-argument
def connected(client):
    # Connected function will be called when the client is connected to Adafruit IO.
    # This is a good place to subscribe to feed changes.  The client parameter
    # passed to this function is the Adafruit IO MQTT client so you can make
    # calls against it easily.
    print("Connected to Adafruit IO!  Listening for DemoFeed changes...")
    # Subscribe to changes on a feed named DemoFeed.
    client.subscribe("DemoFeed")

def subscribe(client, userdata, topic, granted_qos):
    # This method is called when the client subscribes to a new feed.
    print('Subscribed to {0} with QOS level {1}'.format(topic, granted_qos))

def unsubscribe(client, userdata, topic, pid):
    # This method is called when the client unsubscribes from a feed.
    print('Unsubscribed from {0} with PID {1}'.format(topic, pid))

# pylint: disable=unused-argument
def disconnected(client):
    # Disconnected function will be called when the client disconnects.
    print("Disconnected from Adafruit IO!")

# pylint: disable=unused-argument
def message(client, feed_id, payload):
    # Message function will be called when a subscribed feed has a new value.
    # The feed_id parameter identifies the feed, and the payload parameter has
    # the new value.
    print("Feed {0} received new value: {1}".format(feed_id, payload))

# Connect to WiFi
wifi.connect()

# Initialize a new MQTT Client object
mqtt_client = MQTT(
    socket=socket,
    broker="io.adafruit.com",
    username=secrets["aio_username"],
    password=secrets["aio_key"],
    network_manager=wifi
)

# Initialize an Adafruit IO MQTT Client
io = IO_MQTT(mqtt_client)

# Connect the callback methods defined above to Adafruit IO
io.on_connect = connected
io.on_disconnect = disconnected
io.on_subscribe = subscribe
io.on_unsubscribe = unsubscribe
io.on_message = message

# Connect to Adafruit IO
io.connect()

# Below is an example of manually publishing a new  value to Adafruit IO.
last = 0
print("Publishing a new message every 10 seconds...")
while True:
    # Explicitly pump the message loop.
    io.loop()
    # Send a new message every 10 seconds.
    if (time.monotonic() - last) >= 5:
        value = randint(0, 100)
        print("Publishing {0} to DemoFeed.".format(value))
        io.publish("DemoFeed", value)
        last = time.monotonic()

# You can also call loop_blocking if you only want to receive values.
# NOTE: If uncommented, no code below this line will run.
# io.loop_blocking()
```

jerryneedell, 2019-12-28:
I think this is related to this setting https://github.com/adafruit/circuitpython/blob/master/ports/stm32f4/mpconfigport.mk#L16

encode needs MICROPY_CPYTHON_COMPAT but that is only include in the CIRCUITPY_FULL_BUILD

I'll try setting this to CIRCUITPY_FULL_BUILD

hukuzatuna, 2019-12-28:
Thank you for working it, Jerry! Let me know when you'd like me to test and verify, and I'll do a pull/sync/update and build the new firmware.

jerryneedell, 2019-12-28:
I was able to create a workaround by editing adafruit_minimqtt.py tp replace all the .encode calls to use ```bytes(string,"UTF-8")``` or equvalent fro the calls:
here are the diffs:
```
jerryneedell@Ubuntu-Macmini:~/projects/stm32f405/airlift$ diff adafruit_minimqtt.py ~/projects/adafruit_github/Adafruit_CircuitPython_MiniMQTT/
121c121
<         if self.password is not None and len(bytes(password,"utf-8")) > MQTT_TOPIC_LENGTH_LIMIT:
---
>         if self.password is not None and len(password.encode('utf-8')) > MQTT_TOPIC_LENGTH_LIMIT:
355c355
<             msg = str(bytes(msg,"ascii"))
---
>             msg = str(msg).encode('ascii')
357c357
<             msg = str(bytes(msg,"utf-8"))
---
>             msg = str(msg).encode('utf-8')
690c690
<             self._sock.send(bytes(string, "utf-8"))
---
>             self._sock.send(str.encode(string, 'utf-8'))
705c705
<         elif len(bytes(topic,'utf-8')) > MQTT_TOPIC_LENGTH_LIMIT:
---
>         elif len(topic.encode('utf-8')) > MQTT_TOPIC_LENGTH_LIMIT:
```
with this change I can run the adafruit_minimqtt_adafruitio demo:
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 2.0.0-6036-g7387f6092 on 2019-12-28; Feather STM32F405 Express with STM32F405RG
>>> import minimqtt_adafruitio_wifi
Connecting to Adafruit IO...
Connected to Adafruit IO! Listening for topic changes on jerryn/feeds/onoff
Sending photocell value: 0...
Sent!
Sending photocell value: 1...
Sent!
Sending photocell value: 2...
Sent!
Sending photocell value: 3...
Sent!
Sending photocell value: 4...
Sent!
Sending photocell value: 5...
Sent!
Sending photocell value: 6...
Sent!
Sending photocell value: 7...
Sent!
Sending photocell value: 8...
Sent!
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "minimqtt_adafruitio_wifi.py", line 115, in <module>
KeyboardInterrupt: 
>>> 
```

should the minimqtt library be changed or should this be fixed by enabling encode for the stm32f405?

jerryneedell, 2019-12-28:
false hope -- this workaround does not work for the originally posted  issue 
fails with ```Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 2.0.0-6036-g7387f6092 on 2019-12-28; Feather STM32F405 Express with STM32F405RG
>>> 
>>> import minimqtt_demo
Connected to Adafruit IO!  Listening for DemoFeed changes...
Subscribed to jerryn/feeds/DemoFeed with QOS level 0
Publishing a new message every 10 seconds...
Publishing 38 to DemoFeed.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "minimqtt_demo.py", line 121, in <module>
  File "adafruit_io/adafruit_io.py", line 422, in publish
  File "adafruit_minimqtt.py", line 374, in publish
IndexError: bytearray index out of range
>>> 
```  not sure if this just needs more tweaking or is another issue

jerryneedell, 2019-12-28:
ok -- fixed the workaround so let me know if you want me to create a PR to adafruit_minimqtt for this.

hierophect, 2019-12-28:
@jerryneedell CIRCUITPY_FULL_BUILD is a setting directed at the features in py/circuitpy_mpconfig.mk, and was created for new boards such as the STM32 that do not include all features yet. It should probably get revisited to either update existing ports to use it (CXD does not, for instance) or remove it in favor of manually disabling everything that isn't used per port. 

Could you try manually adding the MICROPY_CPYTHON_COMPAT setting to mpconfigport.mk, or moving it from full to minimal build in py/circuitpy_mpconfig.mk?

jerryneedell, 2019-12-28:
@hierophect I made this change
```
jerryneedell@Ubuntu-Macmini:~/circuitpython_jerryn/ports/stm32f4$ git diff
diff --git a/py/circuitpy_mpconfig.h b/py/circuitpy_mpconfig.h
index 932d4f855..ff6b7f4ad 100644
--- a/py/circuitpy_mpconfig.h
+++ b/py/circuitpy_mpconfig.h
@@ -182,7 +182,7 @@ typedef long mp_off_t;
 
 // Remove some lesser-used functionality to make small builds fit.
 #define MICROPY_BUILTIN_METHOD_CHECK_SELF_ARG (CIRCUITPY_FULL_BUILD)
-#define MICROPY_CPYTHON_COMPAT                (CIRCUITPY_FULL_BUILD)
+#define MICROPY_CPYTHON_COMPAT                (CIRCUITPY_MINIMAL_BUILD)
 #define MICROPY_MODULE_WEAK_LINKS             (CIRCUITPY_FULL_BUILD)
 #define MICROPY_PY_ALL_SPECIAL_METHODS        (CIRCUITPY_FULL_BUILD)
 #define MICROPY_PY_BUILTINS_COMPLEX           (CIRCUITPY_FULL_BUILD)
```
and it now works with the minimqtt library -- so MICROPY_CPYTHON_COMPAT can be enabled. It is not clear how to enable it just for this board.

one issue that turned up in the minimqtt library is that is uses microcontroller.cpu.uid  to set a "client id" but for some reason it assumes the length  of microcontroller.cpu.uid is 15 bytes. For the STM32F405 it appears to be only 12. It is an easy fix in the minimqtt library, but I wanted to check to see if there was some reason why there is a different in the size and if it matters to anyone (other than the minimqtt library).  Is ther some convention about the uid length taht the stm32f405 is not following or is it arbitrary?

hierophect, 2019-12-28:
@jerryneedell as you seem to have determined yourself, I don't believe there is any particular standard on length that silicon vendors like ST or Microchip follow when setting the unique ID for a chip.

hierophect, 2019-12-28:
In any case, thanks for determining that setting needs changing! Let me regroup on Monday and I'll issue a PR fix for it. Might end up revamping that whole section as noted earlier.

jerryneedell, 2019-12-28:
> In any case, thanks for determining that setting needs changing! Let me regroup on Monday and I'll issue a PR fix for it. Might end up revamping that whole section as noted earlier.

Glad to help - I learned a lot from this.

jerryneedell, 2020-01-06:
PR #2439 does not appear to have fixed this

```Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.2-90-g776c9b011 on 2020-01-04; Feather STM32F405 Express with STM32F405RG
>>> 
>>> import minimqtt_demo
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "minimqtt_demo.py", line 94, in <module>
  File "adafruit_minimqtt.py", line 120, in __init__
AttributeError: 'str' object has no attribute 'encode'
>>> string = "adsdas"
>>> dir(string)
['count', 'endswith', 'find', 'format', 'index', 'isalpha', 'isdigit', 'islower', 'isspace', 'isupper', 'join', 'lower', 'lstrip', 'replace', 'rfind', 'rindex', 'rsplit', 'rstrip', 'split', 'startswith', 'strip', 'upper']
>>> 
>>> 
```

hierophect, 2020-01-06:
@jerryneedell it might have been that MICROPY_CPYTHON_COMPAT needs to be overridden, not merely defaulted as it is in the PR. #2439 got merged before you had a chance to test it, which wasn't my intention - I don't have a adafruit_minimqtt setup available and thus didn't test the PR myself, I should have made that more clear in the PR text. I'll submit a new PR with a note that it should not be merged until it is working on your setup.

jerryneedell, 2020-01-06:
> @jerryneedell it might have been that MICROPY_CPYTHON_COMPAT needs to be overridden, not merely defaulted as it is in the PR. #2439 got merged before you had a chance to test it, which wasn't my [...]

the easiest way to verify is to do as above  -- just create a string and see if string.encode exits.

hierophect, 2020-01-06:
@jerryneedell I will need more information about your setup to replicate this error - `import minimqtt_demo` does not work with the base library package. Are you working from a guide? If you could provide more information about your exact setup so I can replicate it, that would be ideal.

hierophect, 2020-01-06:
If you give https://github.com/adafruit/circuitpython/pull/2457 a spin, it will hopefully resolve your issue.

jerryneedell, 2020-01-09:
Fixed by #2457

PR #2439 (2019-12-30): STM32: Cpython compatibility flag
Adds the MICROPY_CPYTHON_COMPAT flag to the STM32 port, which does not currently use the FULL_BUILD setting. Resolves #2433. 

I think this might be a good opportunity to discuss the build settings in `py/circuitpy_mpconfig.mk`, since they're getting a little bloated. Right now we have:

- CIRCUITPY_FULL_BUILD (everything on the list)
- CIRCUITPY_SMALL_BUILD (just turns off full build only items)
- CIRCUITPY_DEFAULT_BUILD (everything not in full build)
- CIRCUITPY_MINIMAL_BUILD (turns off default build items)
- CIRCUITPY_ALWAYS_BUILD (items with no HAL compone [... cut]
PR #2457 (2020-01-13): STM32: Cpython compatibility flag 2
A continuation of #2439, which got merged prematurely. Please do not merge until @jerryneedell has had a chance to review. Should resolve #2433  when finished.
closed 2020-01-09 by jerryneedell