adafruit/circuitpython #2435: Inconsistent python alias in py.mk

theodox, 2019-12-29:
It looks like lines 307 and 312 of `py.mk` use the alias `PYTHON` rather than `PYTHON3` which is used later in the file (at 330 and 338).   Evidently this breaks building mpy-cross if your system does not point `python` at the `python3` .  Changing the `PYTHON` references to `PYTHON3` fixed it for me on a completely vanilla, brand-new WSL install of ubuntu 18.04

jerryneedell, 2019-12-29:
Where is this py.mk file you are referring to?  I have not had any issue building mpy-cross on my Linux system with python and python3.
at the top level directory of the repostitory I use:
```make -C mpy-cross/``

tannewt, 2020-01-02:
@theodox Please create a PR with the fix. We'd love to merge it in. Let us know if you need help doing that. Thanks!

dhalbert, 2020-01-15:
Fixed by #2505. Thanks!

closed 2020-01-15 by dhalbert