adafruit/circuitpython #2436: STM32F405 Feather fails to show as Circuit Pyhon device after programming.

rfefx, 2019-12-30:
I have flashed this STM F405 feather numerous times. 
here are my repeated steps: 
install jumper wire between 3.3v and B0. 
insert usb cable. yellow light flashes.
remove jumper wire. 
program following STM32CubeProg steps and receive "File download complete alert"

what i get is the neopixel pulsing green, the yellow led still rapidly flashing, and no circuitpython drive coming up on my computer.

jerryneedell, 2019-12-30:
What version of CIrcuitPython are you using? There is a known issue with 5.0beta2 on the stm32f405. You will need to use the latest build from https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/feather_stm32f405_express/en_US/

rfefx, 2019-12-30:
Thank you very much. Issue Resolved. :)

closed 2019-12-30 by rfefx