adafruit/circuitpython #2437: import adafruit_circuitplayground.express from an imported file fails

dmopalmer, 2019-12-30:
With a file `testme.py` which consists of the single import statement:
```
import adafruit_circuitplayground.express
```
I get a maximum recursion depth error:
```
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.0 on 2019-11-19; Adafruit CircuitPlayground Express with samd21g18
>>> import testme
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "testme.py", line 1, in <module>
  File "adafruit_circuitplayground/express.py", line 699, in <module>
  File "adafruit_circuitplayground/express.py", line 127, in __init__
  File "adafruit_lis3dh.py", line 321, in __init__
  File "adafruit_lis3dh.py", line 100, in __init__
  File "adafruit_lis3dh.py", line 126, in data_rate
  File "adafruit_lis3dh.py", line 300, in _read_register_byte
  File "adafruit_lis3dh.py", line 325, in _read_register
RuntimeError: maximum recursion depth exceeded
>>> 
```

I get this on the latest version 4.1.2 and version 5.0.0-beta.2 as well, although those don't dump the stack trace, but crash and then complain on the next reboot that the stack was corrupted. 

This is with the lib directory empty.  

Doing the same `import adafruit_circuitplayground.express` from the command line  has no problems.
Doing that import from the command line and then `import testme` also works fine.

I tried this on two different Circuit Playground Express hardware units.

dhalbert, 2019-12-30:
So, TODO: Increase CPX stack size a bit more on 4.1.x and 5.x

closed 2020-01-22 by dhalbert

barbudor, 2020-10-22:
Hello,
Just undusted my CPX and upgraded to latest CPY 5.3.1 and I'm facing the exact same problem.
With file `code.py` being
```
from adafruit_circuitplayground import cp
print("hello CPX");
```
everything is fine
But if renaming the file as `test.py` and manually importing from REPL, I get the error :
```
>>> import test
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "test.py", line 1, in <module>
  File "adafruit_circuitplayground/__init__.py", line 29, in <module>
  File "adafruit_circuitplayground/express.py", line 75, in <module>
  File "adafruit_circuitplayground/express.py", line 72, in __init__
  File "adafruit_circuitplayground/circuit_playground_base.py", line 112, in __init__
  File "adafruit_lis3dh.py", line 318, in __init__
  File "adafruit_bus_device/i2c_device.py", line 31, in <module>
RuntimeError: maximum recursion depth exceeded
```

Could you please advise a new stack value to set ? I can't find what is the current value ?
Thanks

barbudor, 2020-10-22:
It's working with `supervisor.set_next_stack_limit(5200)` but not with `5100`

tannewt, 2020-10-22:
@barbudor We don't recommend manually importing from the REPL. Working from `code.py` is enough.

barbudor, 2020-10-23:
Hi @tannewt ,
Allow me to be quite surprised by your comment or I don't understand it well.
I'm used to do unitray testing by loading modules and manually calling functions from the REPL.
If I can't import a module, I'm not sure to understand what the REPL is meant for.
Thanks

tannewt, 2020-10-23:
@barbudor I recommend iterating by modifying `code.py` repeatedly. I'm not personally a fan of the REPL because code is ephemeral. Ultimately, there will always be a limit to what can be imported into the REPL because we have limited resources. When these limits are reached, optimization is needed.

barbudor, 2020-10-23:
Hi @tannewt 
Understood
Thanks

lindes, 2021-04-21:
Hi @tannewt,

Thanks for all you do on this.  I've been watching your streams lately, and really appreciate your efforts.

That said, I do have a bit of a complaint about the response above...

While I can certainly understand a "we don't support this modality of things because of resource limitations on the platform" (a paraphrased restatement of what I think you're fundamentally trying to say), I find "I'm not personally a fan of the REPL" (direct quote) to be rather dismissive, and really, dismissive of what I (and I think many) would consider best practices for programming in general.  REPLs have a long history of being useful in exotic conditions, too...  to wit (from [here](http://www.flownet.com/gat/jpl-lisp.html)):

> (Debugging a program running on a $100M piece of hardware that is 100 million miles away is an interesting experience. Having a read-eval-print loop running on the spacecraft proved invaluable in finding and fixing the problem. The story of the Remote Agent bug is an interesting one in and of itself.)"

I don't pretend that the project I'm working on is anything remotely as significant (or as remote; it's mere centimeters away from the hands that are typing this).  I do, however, think it would "prove invaluable" to be able to import a module that imports cp, specifically so that I can (a) have things be, well, modular (hence wanting the module at all), and (b) not have to type `from adafruit_curcuitplayground import cp` any time I want to debug something (the module I'm using has a 3-letter name, so it's rather easier to import -- or would be, if it was working to do so).

The fact that the code there is ephemeral is immaterial to me.  I use the REPL specifically to try to understand how things work, and I go back to editing my files to do the things I want to ultimately do.  The two modalities solve different problems.  That said, it's useful to have as-easy-as-possible access to as-close-as-possible-to-what-I-would-otherwise-be-using code from within the REPL, for a variety of reasons.

Anyway, I'll carry on with one of the workarounds above (though importing `cp` before my library didn't actually work either... alas[0]... or did it?[1])... and I don't necessarily expect this to change.  I just hope you'll reflect on the fact that a REPL allows things that re-uploading a file does not, and vice versa.  It would be nice if one could do stuff like this and have it work.

Thanks for listening.  I hope the "complaining" nature of this comment isn't too strong in tone.  I really do appreciate everything you've been doing.  I just want things to be the best they can be, and... the above (both not having a fix for this, and the specific piece of the discourse I'm pointing out) seems to me to miss that mark.

Then again, resources (both technical and human) are limited, so... I'm fine with this staying closed.  I simply wanted to present some food for thought, though, on how things might be better.

Best,

- David

[0] I got a `MemoryError: memory allocation failed, allocating 447 bytes`... I don't know if I was already right at the threshold of memory limits before I started refactoring, or what, but I hadn't been getting this when a previous version of essentially the same code was all in `code.py`.  I'll have to figure out if I was close to the limits before refactoring, or what.  Anyway, I'll figure something out... Not trying to report this as a new problem, just letting you know why it didn't work.

[1] If I `import supervisor` and then `supervisor.set_next_stack_limit(5200)`, as mentioned above, at the beginning of my module file, things "Just Work".  So there's that!

dhalbert, 2021-04-21:
@lindes Thank you for your comments. We are very resource-constrained on the CPX, and have to balance stack size vs heap size. Does this still fail with 6.x? We turned on using a heap-based stack frames for Python.

@tannewt I know this is old, but I also use `import test` kind of things when doing some iterations. There are sometimes reasons why I don't want to use `code.py` (for instance, I need to start a clean Saleae trace before importing), but instead import manually from the REPL. If we can slightly increase the stack size (or perhaps figure out how to refactor the library to avoid the deeper stack), I think it would be good to avoid this error when using a common library.

tannewt, 2021-04-21:
@lindes That is how I use the REPL too. I'm not suggesting removing the REPL. This is a weird case where it's out of resources. In that case, I think it's ok to suggest using the code.py if it works still.

@dhalbert I'm totally open to tweaking the stack sizes. Let's see where we land after updating MicroPython. I think a bunch of stuff has changed internally.

lindes, 2021-04-22:
@dhalbert Gladly.  :)

Given that memory error I mentioned at the end, I ended up starting to use mpy-cross for things, and that seems to have fixed things up for me fairly well.  I've still had occasional problems, but not the stack error -- even after removing the `supervisor.set_next_stack_limit` call entirely.  It even works in the REPL!  Yay!

@tannewt fair enough.  Given the experience I had above, I guess my suggestion for future advice is to advise folks to use mpy-cross if they're hitting this... and then if that _still_ fails, then I'm not sure.

Thanks again, both.

PR #2545 (2020-01-22): Increase CPX stack size by 256
Fixes #2437 on circuitpython 5.0.0.