adafruit/circuitpython #2442: Hardfault handler gets triggered on nRF52840 Feather

rhooper, 2019-12-31:
I have been running into an issue where the hard-fault handler gets triggered on the Feather nRF52840.  Unfortunately, the board starts flashing yellow and isn't visible on USB when this happens:  neither as a serial port, nor as a drive.  Pressing reset is required.  

I tried disabling the bluetooth code.
I tried 5.0.0b2
I tried master
I tried new-pixelbuf-api without bluetooth code.
I tried not using pixelbuf at all (just neopixel).
I tired without USB connected.

The same code on the Feather M4 (except the bluetooth part) works fine and is still running hours later.

The NRF board always eventually hardfaults.  All tests are running the animations.  Using pixelbuf speeds up the fault.  Using the classes that slice pixelbufs cause faster faults.

Tracing with the JLink was inconclusive... The hardfault was happening from an unknown address.

TODO: Test on a fresh Feather NRF board.

dhalbert, 2019-12-31:
Did you set a breakpoint on `HardFault_Handler`?

rhooper, 2019-12-31:
Even on a fresh board, it faulted.

I did, it didn't trigger.  Interrupting execution showed it in the NORDIC_SOFT_DEVICE_ASSERT safe mode.  I set a breakpoint on softdevice_assert_handler.  See below.

```
(gdb) break HardFault_Handler
Breakpoint 1 at 0x43b16: file supervisor/port.c, line 168.
(gdb) run
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /Users/rhooper/Devel/circuitpython/ports/nrf/build-feather_nrf52840_express/firmware.elf
^C
Program received signal SIGTRAP, Trace/breakpoint trap.
[Switching to Thread 57005]
0x0004377c in run_code_py (safe_mode=safe_mode@entry=NORDIC_SOFT_DEVICE_ASSERT) at ../../main.c:301
301	        if (!refreshed_epaper_display) {
(gdb) bt
#0  0x0004377c in run_code_py (safe_mode=safe_mode@entry=NORDIC_SOFT_DEVICE_ASSERT) at ../../main.c:301
#1  0x0004398e in main () at ../../main.c:462
```

```
Breakpoint 2, softdevice_assert_handler (id=1, pc=87328, info=0) at common-hal/_bleio/Adapter.c:66
66	STATIC void softdevice_assert_handler(uint32_t id, uint32_t pc, uint32_t info) {
(gdb) bt
#0  softdevice_assert_handler (id=1, pc=87328, info=0) at common-hal/_bleio/Adapter.c:66
#1  0x00024a44 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

dhalbert, 2019-12-31:
One reason this can happen is if the softdevice detects that it has missed a BLE timing window. This is often due to something else taking too long with interrupts off. I did search for this `pc` value, but no one has reported that.

It could be that `neopixel_write` is taking too long somehow. Could you try this specifying DotStars instead? You don't have to actually connect anything, just try the same code.

rhooper, 2019-12-31:
That would make sense - `neopixel_write` is called the majority of the execution time in that particular code. Is there some way to make sure BLE timing windows are not missed?  The dotstar test is ongoing.  I'll post if it ever crashes to the NORDIC_SOFT_DEVICE_ASSERT.

Is there a better way to handle NORDIC_SOFT_DEVICE_ASSERT?

dhalbert, 2019-12-31:
>Is there a better way to handle NORDIC_SOFT_DEVICE_ASSERT?

It should never happen. It means something bad has happened that it can't recover from.

tannewt, 2020-01-02:
Any idea what is turning off interrupts? I thought the nRF neopixel code used a peripheral to do it and didn't mess with interrupts.

rhooper, 2020-01-02:
Not sure.  https://github.com/adafruit/circuitpython/blob/master/ports/nrf/common-hal/neopixel_write/__init__.c#L229 might be if DMA isn't used - I'll check with a JLink later.

rhooper, 2020-01-03:
Is setting a breakpoint on `__disable_irq` the right way to catch what might be disabling interrupts?

rhooper, 2020-01-03:
I saw a `__disable_irq()` during `neopixel_write`.  Here's the traceback.
```
#0  common_hal_neopixel_write (digitalinout=digitalinout@entry=0x20008790, pixels=0x20009eb0 "\023", numBytes=960)
    at common-hal/neopixel_write/__init__.c:229
#1  0x0005e79c in neopixel_write_neopixel_write_ (digitalinout_obj=0x20008790, buf=<optimized out>)
    at ../../shared-bindings/neopixel_write/__init__.c:74
#2  0x000364c6 in fun_builtin_2_call (self_in=0x82e84 <neopixel_write_neopixel_write_obj>, n_args=<optimized out>,
    n_kw=<optimized out>, args=0x2003f9a0) at ../../py/objfun.c:84
#3  0x00031dbe in mp_call_function_n_kw (fun_in=0x82e84 <neopixel_write_neopixel_write_obj>, n_args=2, n_kw=0,
    args=args@entry=0x2003f9a0) at ../../py/runtime.c:622
#4  0x0003f6ba in mp_execute_bytecode (code_state=code_state@entry=0x2003f988, inject_exc=<optimized out>, inject_exc@entry=0x0)
    at ../../py/vm.c:932
#5  0x00036620 in fun_bc_call (self_in=0x200385f0, n_args=<optimized out>, n_kw=204223, args=0x2003fab0)
    at ../../py/objfun.c:284
#6  0x00031dbe in mp_call_function_n_kw (fun_in=0x200385f0, n_args=1, n_kw=0, args=args@entry=0x2003fab0)
    at ../../py/runtime.c:622
#7  0x00031e88 in mp_call_method_n_kw (n_args=<optimized out>, n_kw=<optimized out>, args=args@entry=0x2003faac)
    at ../../py/runtime.c:637
#8  0x0003f706 in mp_execute_bytecode (code_state=code_state@entry=0x2003fa98, inject_exc=<optimized out>, inject_exc@entry=0x0)
    at ../../py/vm.c:1015
#9  0x00036620 in fun_bc_call (self_in=0x200373d0, n_args=<optimized out>, n_kw=204223, args=0x20025d88)
    at ../../py/objfun.c:284
#10 0x00031dbe in mp_call_function_n_kw (fun_in=0x200373d0, n_args=1, n_kw=0, args=args@entry=0x20025d88)
    at ../../py/runtime.c:622
#11 0x00031e88 in mp_call_method_n_kw (n_args=<optimized out>, n_kw=<optimized out>, args=args@entry=0x20025d84)
    at ../../py/runtime.c:637
#12 0x0003f706 in mp_execute_bytecode (code_state=code_state@entry=0x20025d70, inject_exc=<optimized out>, inject_exc@entry=0x0)
    at ../../py/vm.c:1015
#13 0x00036620 in fun_bc_call (self_in=0x20035fe0, n_args=<optimized out>, n_kw=204223, args=0x2003fc98)
    at ../../py/objfun.c:284
#14 0x00031dbe in mp_call_function_n_kw (fun_in=0x20035fe0, n_args=1, n_kw=0, args=args@entry=0x2003fc98)
    at ../../py/runtime.c:622
#15 0x00031e88 in mp_call_method_n_kw (n_args=<optimized out>, n_kw=<optimized out>, args=args@entry=0x2003fc94)
    at ../../py/runtime.c:637
#16 0x0003f706 in mp_execute_bytecode (code_state=code_state@entry=0x2003fc80, inject_exc=<optimized out>, inject_exc@entry=0x0)
    at ../../py/vm.c:1015
#17 0x00036620 in fun_bc_call (self_in=0x200375b0, n_args=<optimized out>, n_kw=204223, args=0x2003fdb8)
    at ../../py/objfun.c:284
#18 0x00031dbe in mp_call_function_n_kw (fun_in=0x200375b0, n_args=1, n_kw=0, args=args@entry=0x2003fdb8)
    at ../../py/runtime.c:622
#19 0x00031e88 in mp_call_method_n_kw (n_args=<optimized out>, n_kw=<optimized out>, args=args@entry=0x2003fdb4)
---Type <return> to continue, or q <return> to quit---
    at ../../py/runtime.c:637
#20 0x0003f706 in mp_execute_bytecode (code_state=code_state@entry=0x2003fda0, inject_exc=<optimized out>, inject_exc@entry=0x0)
    at ../../py/vm.c:1015
#21 0x00036620 in fun_bc_call (self_in=0x20035b30, n_args=<optimized out>, n_kw=204223, args=0x200086c8)
    at ../../py/objfun.c:284
#22 0x00031dbe in mp_call_function_n_kw (fun_in=0x20035b30, n_args=1, n_kw=0, args=args@entry=0x200086c8)
    at ../../py/runtime.c:622
#23 0x00031e88 in mp_call_method_n_kw (n_args=<optimized out>, n_kw=<optimized out>, args=args@entry=0x200086c4)
    at ../../py/runtime.c:637
#24 0x0003f706 in mp_execute_bytecode (code_state=code_state@entry=0x200086b0, inject_exc=<optimized out>, inject_exc@entry=0x0)
    at ../../py/vm.c:1015
#25 0x00036620 in fun_bc_call (self_in=0x20038b30, n_args=<optimized out>, n_kw=204223, args=0x0) at ../../py/objfun.c:284
#26 0x00031dbe in mp_call_function_n_kw (fun_in=fun_in@entry=0x20038b30, n_args=n_args@entry=0, n_kw=n_kw@entry=0,
    args=args@entry=0x0) at ../../py/runtime.c:622
#27 0x00031de8 in mp_call_function_0 (fun=fun@entry=0x20038b30) at ../../py/runtime.c:596
#28 0x0004ee42 in parse_compile_execute (source=source@entry=0x78018, input_kind=input_kind@entry=MP_PARSE_FILE_INPUT,
    exec_flags=exec_flags@entry=32, result=result@entry=0x2003ffbc) at ../../lib/utils/pyexec.c:114
#29 0x0004f180 in pyexec_file (filename=filename@entry=0x78018 "code.py", result=result@entry=0x2003ffbc)
    at ../../lib/utils/pyexec.c:529
#30 0x000435ca in maybe_run_list (filenames=filenames@entry=0x20004038 <supported_filenames>,
    exec_result=exec_result@entry=0x2003ffbc) at ../../main.c:183
#31 0x0004369a in run_code_py (safe_mode=safe_mode@entry=NO_SAFE_MODE) at ../../main.c:241
#32 0x0004398e in main () at ../../main.c:462
```

dhalbert, 2020-01-03:
Based on that traceback, the nRF `neopixel_write` implementation is falling back to a busy-waiting NeoPixel write implementation:, which will mess up the BLE timing. That will happen if it can't get a PWM peripheral. So `find_free_pwm()` is returning `NULL` for some reason. Maybe the PWM peripheral is not reset properly and appears to still be in use even after it's freed after the neopixel write.

I think the correct thing to do instead is if we can't get a PWM peripheral, then check to see if the softdevice is enabled, and if it is, then give up and throw an exception instead of just falling back to the non-PWM implementation. Here is how to test if the SD is enabled:
```c
        uint8_t sd_en = 0;
        (void) sd_softdevice_is_enabled(&sd_en);
        if (sd_en) {
...
```

https://github.com/adafruit/circuitpython/blob/32b82032a1af88314aa260665a4bfd8c4b7a6988/ports/nrf/common-hal/neopixel_write/__init__.c#L221-L229

tannewt, 2020-01-03:
I'll take a look at the pwm reset logic. It sounds similar to the issue I found with UART reset.

Is SD-less CP something we actually want to support? I assume BLE will be on most of the time.

dhalbert, 2020-01-03:
> Is SD-less CP something we actually want to support? I assume BLE will be on most of the time.

If we can do a deep sleep with SD on, then it's OK to assume it's on. I think before you added the CircuitPython service, the SD was enabled only if `_bleio` was imported.

dhalbert, 2020-01-03:
And it can still be disabled (`Adapter.disable`), so yes, I think we should still check whether the SD is enabled or not.

tannewt, 2020-01-03:
I don't actually think this is a reset issue. The buffer for PWMing the neopixels is huge `totalMem = numBytes*8*2+(2*2)` and in the stack trace `numBytes` is 960 for a total of 15364 bytes.

I'll have it raise an OOM if the SD is on.

rhooper, 2020-01-03:
@tannewt Right, the more pixels, the more likely this is to be triggered (but even doing just 32 Neopixels was enough to eventually trigger the fault).  During that test, I had the pixel count set to either 240 or 320 (I forget which now).   I wonder why that eventually becomes an issue?  This issue was happening whether or not I had adafruit_ble imported.  

What further testing (and/or breakpoints or other gdb commands) can I do to provide more context?

tannewt, 2020-01-04:
@rhooper give it a try now that I have it throwing a memoryexception. I have a feeling it'll fall into the "memory fragmentation sucks" bucket which doesn't have an easy fix. One option would be for neopixel_write to hold onto its memory indefinitely and reuse and expand it as needed. That could reduce heap churn.

deshipu, 2020-01-04:
Another option would be to make it accept an optional pre-allocated buffer, like the audioio does now. This way the user can control the lifetime of the buffer if it's needed.

rhooper, 2020-01-04:
@deshipu That was going to be my suggestion, rather than having neopixel_write hold onto memory.

rhooper, 2020-01-04:
Upon further thought, this is a port-specific behaviour.  Having neopixel_write take a buffer just for one port isn't a great idea.  The suggestion by @tannewt to reuse (or grow) a buffer is a good one and should be easy to implement.

A future optimization might be to free the buffer (on NRF) if garbage collection is called manually from gc.collect?

dhalbert, 2020-01-04:
The possibility of a using pre-allocated buffer could be hidden in the calling libraries. `neopixel_write` could report this possibility with a property or similar.

deshipu, 2020-01-04:
I wonder, do we really need to fit the whole thing in one buffer? Would it be possible to use two small buffers for double-buffering back to back, and fill one while the other is playing out? This way it could handle any number of pixels with constant memory.

rhooper, 2020-01-04:
Might be able to use a single smaller static buffer - though I'm not sure if the timing would remain accurate.  That said, it would be easy to test if a fixed buffer works, as DMA mode still has a wait loop.  https://github.com/adafruit/circuitpython/blob/master/ports/nrf/common-hal/neopixel_write/__init__.c#L213

It isn't clear whether use of two sequences and NRF_PWM_TASK_SEQSTART1, NRF_PWM_EVENT_SEQEND1 would be helpful.

rhooper, 2020-01-04:
This is still running ... ~11 hours later.

dhalbert, 2020-02-22:
Fixed by #2648.

closed 2020-02-22 by dhalbert