adafruit/circuitpython #2443: CurcuitPython hard crash report

noamziv, 2020-01-01:
Messege: Crash into the HardFault_Handler

[code.zip](https://github.com/adafruit/circuitpython/files/4014443/code.zip)

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

dhalbert, 2020-01-01:
Which board and which version of CircuitPython is this?

noamziv, 2020-01-01:
Trinket M0
adafruit-circuitpython-trinket_m0-en_US-4.1.0

noamziv, 2020-01-01:
using the I2c drivers (adafruit_bus_device) from
adafruit-circuitpython-bundle-4.x-mpy-20191123

dhalbert, 2020-01-01:
```python3
            except:
                # Handle IO exception by waiting it out
                time.sleep(0.010)
                #print("Exception")
```

What exception are you getting above, on which line? Could you remove the handler temporarily so we can see the backtrace? Also, have you tried updating to 5.0.0-beta.2 and also updating the librariies? It seems like the FX29 may be pathological in some way.

noamziv, 2020-01-01:
The exception was:
'OSError: [Errno 5] EIO\r\n'

I do not have line info but I believe it occurs during an I2C read when the device is not ready [ device.readinto() ]

Unfortunately the crash occurs irregularly and almost randomly.  I have the code running on multiple "systems" and at best (from debug point of view), a crash occurs after several hours, and at worst never.  This one happened overnight.

I have not tried updating as of yet.

Is there a way to force an automatic restart/reboot when a fault occurs?  That would be a quick "fix" for my use case, while we continue to investigate.  Right now I have to manually reset the trinket (by either pressing the reset button or cycling power)

And the FX29 is indeed an interesting device.

dhalbert, 2020-01-01:
You can call `microcontroller.reset()` to do the equivalent of pressing the reset button.

Do you need 100Hz update? what happens if you lengthen the `sleep()` in the main loop. I see that the datasheet says it can take 3 msecs or 8.4msecs to get a reading (at "4 MHz" -- I'm not sure what clock frequency it's talking about there, because the max I2C freq is 400 kHz).

Also there is a typo in your test program:
```
                    FaultCount = FalutCount + 1
```

dhalbert, 2020-01-01:
All of this does not excuse the crash, which should not happen, though there are some devices that really confuse the I2C peripheral on some chips.

dhalbert, 2020-01-01:
Could you give us a part number or purchase link for the particular FX29 you are using? It may be worth ordering one and trying it out.

noamziv, 2020-01-01:
Thanks for catching the Fault with my Fault Counter :)

I do need the 100Hz Update. This should be achievable - I tested without the delay and was getting continuous updates at a much higher rate without any errors.

And the FX29 documentation certainly leaves room for improvement.  Here is the documentation for the I2C controller that is in it.

[IDT_ZSC31015-Datasheet_DST_20161114.pdf](https://github.com/adafruit/circuitpython/files/4014484/IDT_ZSC31015-Datasheet_DST_20161114.pdf)

The FX29 I'm using is: FX29J0-040B-0100-L

Thank you for the help!

noamziv, 2020-01-01:
Or Digikey link
https://www.digikey.com/products/en?keywords=FX292X-100A-0100-L

dhalbert, 2020-01-01:
I assume you have I2C pullups somewhere? Are they 10K each, and are they located at the end of the FX29 wires (which seem longish?). Is there any special stressing going on with the load cell when it crashes, or is it just sitting quietly?

I am just asking all this to make sure we can duplicate your setup. Thanks.

noamziv, 2020-01-01:
We are using 4.7K pullups.

The wires we ended up with are shorter - we are deploying the 40mm length.

Crashes are uncorrelated with pressure, or changes in pressure.  Usually the crash happens when there is a constant (low) pressure.  My overnight failure happened on the bench, with no pressure at all.

I appreciate the questions, and the help.

In my code I plan to replace the time.sleep() statement with the microcontroller.reset() in the exception handler.  Is there an easy way for me to cause an exception "on demand"?  

Thanks again!

dhalbert, 2020-01-01:
To cause an exception, you just `raise` it: `raise OSError`, for instance.

The exact part number you gave, FX29J0-040B-0100-L, does not seem to be stocked anywhere. The "J" says I2C without sleep, and no one stocks J models. https://te.com does not list that as a regular item either, though the datasheet implies you could order it (which I guess you did). See below.

So this looks harder to reproduce on our bench. We could get the I2C controller and try that. But do you have a J-Link and/or one of our boards with an SWD controller? You could set a breakpoint to catch the hard fault and we might try to debug this remotely. Of course, that's a lot of work, and if your final product doesn't use CircuitPython anyway, it may not be worth it.

[image]

noamziv, 2020-01-01:
I believe the sensor is new - which is why we ended up ordering the parts digikey had in stock at the time rather than the ones we originally wanted..   And that means we got the K version (with sleep) rather than the J version (which we wanted). 

I don't have the setup you describe - so remote debugging is not in the cards at this point.  Sorry, but thanks for the offer.

dhalbert, 2020-01-01:
So is the one you actually have in hand a FX29K0-something? DigiKey has none in stock, but Mouser has these, which are pretty similar (load range, cable length, and connector vary). I could get one of those: https://www.mouser.com/TE-Connectivity/Sensors/Force-Sensors-Load-Cells/_/N-deui5Z1yzvvqx?P=1z0s0dsZ1yzs6ii&Keyword=29k0&FS=True

dhalbert, 2020-01-01:
The J vs K sleep difference might be significant and might be confusing the Atmel I2C driver.

noamziv, 2020-01-01:
This is the correct sensor.  The "sleep" feature may indeed be the cause or part of the cause.
Unfortunately, the IDT controller referenced above is programmable only during manufacturing and not during operation - so sleep can not be disabled once a part is in your hands.

All of the sensors are mechanically identical
The load range is determined by a resistor so it should not affect communications
The cable length is short enough so that it should not be a factor
We are using the connectors, but we also cut them off for our prototyping, so they too are not likely to be part of the issue.

so it's back to Sleep and the I2C controller

dhalbert, 2020-01-01:
One more question: the ZSC31015 is a OneWire device, not I2C, so is there another microcontroller that's converting the OneWire to I2C? I would guess so. So whatever firmware is running in that microcontroller might be flaky. Have you talked to TE about this?

noamziv, 2020-01-01:
TE supplied me the ZSC document after a long quest to get info beyond what is found in the data sheet.
I don't hold much hope of getting more useful info beyond what I have..
And so, no.  At least not yet.

dhalbert, 2020-01-15:
The long-term issue to fix here is to prevent a hard fault.

Artem-B, 2020-11-12:
> And the FX29 documentation certainly leaves room for improvement. Here is the documentation for the I2C controller that is in it.
> 
> [IDT_ZSC31015-Datasheet_DST_20161114.pdf](https://github.com/adafruit/circuitpython/files/4014484/IDT_ZSC31015-Datasheet_DST_20161114.pdf)

I think the actual controller inside is [ZSC31014](https://www.idt.com/us/en/document/dst/zsc31014-datasheet).
The diagrams in the FX29 appear to be a slightly photoshopped ones from the ZSC31014 datasheet.
ZSC31014 commands also match the ones used in the TE's app note ["Changing slave i2c address"](https://www.te.com/commerce/DocumentDelivery/DDEController?Action=showdoc&DocId=Specification+Or+Standard%7FMS45x5DO_Changing_I2C%7FA%7Fpdf%7FEnglish%7FENG_SS_MS45x5DO_Changing_I2C_A.pdf%7FCAT-BLPS0001)

The fact that it's possible to change I2C address implies that the EEPROM is **not** locked and it should be possible to tweak various settings. Clock, sleep mode, sample rate -- they all are in the same EEPROM.

kapcem, 2021-07-08:
> I think the actual controller inside is [ZSC31014](https://www.idt.com/us/en/document/dst/zsc31014-datasheet).
> The diagrams in the FX29 appear to be a slightly photoshopped ones from the ZSC31014 [...]

Did you tried the code and successfully changed  the i2c address?

Artem-B, 2021-07-09:
I didn't try to change the address, but I did successfully change the sleep mode. The eeprom **is** writable.

I've got rough circuitpython code for accessing FX29, including putting it in command mode and reading/writing eeprom.
The code assumes that the sensor is powered via one of the I/O pins.
https://github.com/Artem-B/g29-lc-brake/blob/9a248062c8dd1c18d630049240bef1b498b4bc5b/code.py#L40