adafruit/circuitpython #2451: Request VID/PID for MIMXRT10XX board

arturo182, 2020-01-05:
I would like to request PIDs for 2 of my i.MX RT Feathers: 1011 and 1062, and for the MIMXRT1010_EVK board, so 3 in total.

Thanks!

ladyada, 2020-01-05:
> iMX RT1011 Feather VID 0x239A   PID 0x0073  # bootloader
>                                 PID 0x8073  # arduino
>                                 PID 0x8074  # circuitpython
> 
> iMX RT1062 Feather VID 0x239A  PID 0x0075  # bootloader
>                                PID 0x8075  # arduino
>                                PID 0x8076  # circuitpython
> 
> MIMXRT1010_EVK     VID 0x239A   PID 0x0077  # bootloader
>                                 PID 0x8077  # arduino
>                                 PID 0x8078  # circuitpython
>

closed 2020-01-05 by ladyada

arturo182, 2020-01-05:
Thanks a bunch!