adafruit/circuitpython #2465: Some class inheritance broken in 5.0.0 beta 3

caternuson, 2020-01-09:
Re this thread:
https://forums.adafruit.com/viewtopic.php?f=58&t=160746

Recreated underlying issue with a simple REPL test:

**5.0.0 beta 2**
```python
Adafruit CircuitPython 5.0.0-beta.2 on 2019-12-20; Adafruit Circuit Playground Bluefruit with nRF52840
>>> class Foo(tuple):
...     pass
... 
>>> foo = Foo((1,2))
>>> foo
(1, 2)
>>> foo[0]
1
>>> 
```

**5.0.0 beta 3**
```python
Adafruit CircuitPython 5.0.0-beta.3 on 2020-01-08; Adafruit Circuit Playground Bluefruit with nRF52840
>>> class Foo(tuple):
...     pass
... 
>>> foo = Foo((1,2))
>>> foo
(1, 2)
>>> foo[0]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'Foo' object is not subscriptable
>>> 
```

dhalbert, 2020-01-09:
This may be related to changes made to support `_pixelbuf`. Pinging @rhooper.

rhooper, 2020-01-09:
Yeah, it would be.

I'll have a look to see if this can be fixed in tuple_subscr, or if we need to revert.

@tannewt I think we should consider switching back to the _subscr signature change approach.  I believe I can get that done quickly, and it should restore the expected behaviour, but allow classes like Pixelbuf to interact with the native base during subscr calls.

We'll certainly want to add this as a test case.

tannewt, 2020-01-09:
I'd rather work forwards and fix this issue. I think it'll be better in the long run if native code has access to the subclass objects.

dhalbert, 2020-01-15:
Fixed by #2504.

closed 2020-01-15 by dhalbert

caternuson, 2020-01-15:
Nice!