adafruit/circuitpython #2476: mimxrt10xx: Verify and possibly fix neopixel_write

arturo182, 2020-01-09:
Copied from stm32, seems a bit buggy, only works sometimes (sets the wrong color when not working).

hierophect, 2020-01-09:
I'd like to take a look at this as part of a revamp I want to do to the STM32 version.

arturo182, 2020-01-10:
I only tested by looking at the neopixel and checking if the colors change correctly, I would like to actually connect a scope on the line to see what is really happening there, but not sure when I'll get time to do this.

hierophect, 2020-01-10:
@arturo182 when it goes wrong, what happens? You said wrong color, does that mean "white" or "off" or actually literally the wrong color but with normal brightness and stuff?

arturo182, 2020-01-10:
What I've been seeing is that if I try to set pure red I get sometimes orange, sometimes yellow, sometimes white, and if I keep retrying, eventually it will be red.

hierophect, 2020-01-12:
Do you have a Saleae? Hands down the easiest tool for this kind of monitoring that I've used. Scope will also show it but the Saleae is just so easy to set up and read with.

arturo182, 2020-01-12:
I have a Chinese clone that works with the Saleae UI and I really like it, I was just worried if it's fast enough, also it's digital-only so if there's some weirdness on the line, I would miss it.

PR #2740 (2020-03-31): mimxrt10xx: Fix neopixel_write
Looks like the timing issues were caused by the code being in flash, moving it to ITCM and making sure no flash functions are called in the timing-critical section fixed the issue :)

Closes #2476

[image]
closed 2020-03-31 by tannewt