adafruit/circuitpython #2483: mimxrt10xx: Verify the busio module

arturo182, 2020-01-09:
Just run sanity tests on SPI, I2C and UART, I probably missed something or there's some corner case.

ladyada, 2020-01-10:
i2c scanning does not seem to find write-only/read-only devices (?):
SSD1306 OLED, and GTOP I2C GPS are not found on bus

ladyada, 2020-01-11:
ahh default i2c frequency for scanning (at least) is 1MHz, should be 100khz to allow all things to respond. many devices dont work at 1Mhz :) for now, im testing with
`i2c = busio.I2C(board.SCL0, board.SDA0, frequency=100000)`

ladyada, 2020-01-11:
SPI freq is 2x faster than expected, setting it to 4MHz -> 8MHz clock speed

arturo182, 2020-01-11:
This is all great info, thanks :)

mubes, 2020-02-26:
SPI Frequency is corrected in https://github.com/adafruit/circuitpython/pull/2658

mubes, 2020-02-27:
I2C Frequency is corrected in #2659 , given that UART was also fixed recently I think it's worth re-testing this lot to see if the ticket can be closed....note that the _requested_ I2C frequency is 400KHz.

There remains a lockup under punishing conditions, but that's a different issue.

arturo182, 2020-02-27:
I need more than 24h in a day :D

arturo182, 2020-02-28:
One more patch by @mubes here https://github.com/adafruit/MIMXRT10xx_SDK/pull/1

Should fix the scan issues.

PR #2964 (2020-05-29): mimxrt10xx: Busio cleanup and bugfixes
This PR issues a general cleanup of the BusIO module to account for recent changes to the pin claiming structure, and issues fixes to a number of bugs and other minor feature requests. All changes are being stress-tested across I2C, SPI, UART, and DigitalIO to hit a baseline level of guaranteed functionality. 

It is not currently ready for merge as I'm still tangling with a number of serious issues during testing, including an ugly startup hang, but I'm opening it up to style review and other concerns. 

Resolves: #2485
Resolves: #2483
Resolves: #2480
Re [... cut]
closed 2020-05-29 by tannewt