adafruit/circuitpython #2487: mimxrt10xx: Fix compiler errors

arturo182, 2020-01-09:
Some of them are from the SDK and I don't know how close to original we want to keep it.

hierophect, 2020-05-15:
Current errors are:
```
../../lib/tinyusb/src/portable/nxp/transdimension/dcd_transdimension.c: In function 'dcd_edpt_xfer':
../../lib/tinyusb/src/portable/nxp/transdimension/dcd_transdimension.c:481:33: warning: cast increases required alignment of target type [-Wcast-align]
  481 |   CleanInvalidateDCache_by_Addr((uint32_t*) buffer, total_bytes + 31);
      |                                 ^
sdk/devices/MIMXRT1011/drivers/fsl_clock.c:24:6: warning: "__ARMVFP__" is not defined, evaluates to 0 [-Wundef]
   24 | #if (__ARMVFP__ >= __ARMFPV5__) && \
      |      ^~~~~~~~~~
sdk/devices/MIMXRT1011/drivers/fsl_clock.c:24:20: warning: "__ARMFPV5__" is not defined, evaluates to 0 [-Wundef]
   24 | #if (__ARMVFP__ >= __ARMFPV5__) && \
      |                    ^~~~~~~~~~~
```
NRF and STM32 both implement CFLAGS to turn these off, since their SDKs also make liberal use of undefined preprocessor variables and casts that change alignment: 
```
CFLAGS += -Wno-undef -Wno-shadow -Wno-cast-align
```
These should be introduced alongside `-Werror` once the full baseline of features are in.

hierophect, 2020-05-15:
the usb one I guess we should ask @hathach about

hathach, 2020-05-17:
@hierophect PR is on the way https://github.com/hathach/tinyusb/pull/411

commit 0b22a939 (2021-03-18): mimxrt: Quiet some warnings down in the sdk

Closes #2487
closed 2021-03-18 by jepler