adafruit/circuitpython #2489: mimxrt10xx: Cleanup and finish PWMOut

arturo182, 2020-01-09:
Right now it's more of a proof-of-concept.

yimidi-kr, 2020-04-28:
Still PWM is not supported. Library is supported when use pin 0-9. But, Output is not generated!!!
Please check this issue^^ I tested on version 5.2.0
Thank you everybody... 
I want to see you version 6.0.0 ^^

siddacious, 2020-04-28:
@yimidi-kr please do not spam the same comment on several repos.

ajordan5, 2021-02-15:
Is there any way to get PWM working on any pins at this point? Currently working on a senior capstone project that decided to go with Circuitpython on our teensy, but only now realized that PWM isn't working. I don't get any errors, just no output. Thanks for your help!

hierophect, 2021-02-15:
@ajordan5 it's on my todo list but not at the top. PWM is currently unsupported, the existing code is really just a placeholder. Would it be practical for your senior project to switch over to another chip that's out of beta and in our `stable` category? A SAMD51 or STM32F405 will have all common modules implemented and any Circuitpython code for the i.MX should be portable to them.

ajordan5, 2021-02-15:
@hierophect Yes switching to a chip that has full support is kind of where we were leaning anyway. Looks like that is the way to go.

hierophect, 2021-02-15:
@ajordan5 sounds good, the Feather F405 or Feather M4 Express might fit your needs. i.MX support should be expanded out of the alpha state sometime this spring but it's taking a backseat to other priorities right now.

jepler, 2021-04-02:
Basic PWM now works, and has been tested on the MIMXRT1010-EVK and a prototype Metro M7 1011: #4530

closed 2021-04-02 by jepler