adafruit/circuitpython #2490: mimxrt10xx: Stress-test the flash

arturo182, 2020-01-09:
I've seen an issue on samd where the external flash can get corrupted and either go into read-only mode or totally crap out. 
We should check if this can happen in the mimxrt10xx port too because since the FW and fatfs are on the same external flash, the risk is higher.

hierophect, 2020-05-19:
@tannewt I'm wondering how I would go about doing this. Any suggestions?

arturo182, 2020-05-19:
Hard to say TBH, maybe copying and removing files to and from the CIRCUITPY drive, also resetting while doing that stuff. Trying to fill the drive while doing other operations. Removing without unmounting, removing while something is copied over. Stuff like that.

hierophect, 2020-05-20:
@arturo182 is the risk that the chip becomes unrecoverably bricked? Or is some essential information retained so it can be re-programmed over GDB or the DAP?

Todo: 

- [ ] test drive overfill (fill all space)
- [ ] save file at unacceptable length
- [ ] yank drive while writing
- [ ] increase and decrease file size many times

arturo182, 2020-05-20:
I guess it depends where the corruption occurs, the way I see it there are 3 areas, best to worst:
- the fatfs gets corrupted - you can erase it from CPY and you're good
- the CPY FW gets corrupted - you can reflash using tinyuf2
- the tinyuf2 bootloader gets corrupted - you have to reflash using the NXP USB ROM loader

I think the chance that the chip itself is bricked is quite low, never seen that happen.

tannewt, 2020-05-20:
I wouldn't prioritize this issue. It's too vague to replicate and debug. I also don't see this issue with SAMD so I'm not sure of the background.

So, my vote is to close but I'll leave it up to you two.

hierophect, 2020-05-20:
I ran the basic tests and had no problems. Compared to most, the amount of flash on the EVKs is so monstrous that it's hard to even get to edge cases. It's possible something might crop up on the smaller feathers but I don't see any evidence of trouble so far - we can keep an eye on it but I'll close for now.

closed 2020-05-20 by hierophect