adafruit/circuitpython #2500: mimxrt10xx: Utilize the Touch Screen peripheral in RT1062

arturo182, 2020-01-09:
The RT1062 has a TSC peripheral, could be useful!

From datasheet:

> The controller utilizes ADC hardware trigger function and control switches in touch screen analogue block. The controller only supports 4-wire of 5-wire screen touch modes.

kmatch98, 2021-03-26:
Looking at the datasheet for the RT1062, it looks like it includes a touch screen controller for resistive touch screens:

[image]

tannewt, 2021-03-29:
I doubt we'll add a separate API for this. I think these touch screens can likely be read already with AnalogIn.

ladyada, 2021-04-04:
closing cause these aren't complex devices, and its better to use a common analogin-using API than a special peripheral!

closed 2021-04-04 by ladyada