adafruit/circuitpython #2503: 'Vec2D' object is not subscriptable

reversefold, 2020-01-11:
I am opening this issue here as the error seems like it's a problem with the runtime, not the turtle library.

When running `turtle_square01.py` from https://learn.adafruit.com/turtle-graphics-gizmo/turtle-graphics-on-gizmo I get the following traceback:
```
Traceback (most recent call last):
  File "code.py", line 61, in <module>
  File "/lib/adafruit_turtle.py", line 218, in forward
TypeError: 'Vec2D' object is not subscriptable
```
This is on a Circuit Playground Bluefruit with the TFT Gizmo. I got the same error with both `adafruit-circuitpython-circuitplayground_bluefruit-en_US-5.0.0-beta.3.uf2` and `adafruit-circuitpython-circuitplayground_bluefruit-en_US-20200110-6ea1785.uf2`.

Doing something similar with an extended tuple in CPython 3.7.2 on my laptop works fine so I suspect this is a problem with the runtime.

A simple workaround for now is to switch Vec2D to a has-a relationship instead of is-a:
```
class Vec2D(object):
    def __init__(self, x, y):
        self.tup = (x, y)

    def __getitem__(self, i):
        return self.tup[i]
```

dhalbert, 2020-01-11:
There was a regression in 5.0.0-beta.3 affecting subscripting: #2465. If you are using beta.3 try beta.2 and see if the problem goes away. Thanks.

dastels, 2020-01-11:
I agree with @reversefold that a has-a relationship is preferable in general. I believe I ported that code over from the cpython turtle implementation. Silver lining is that it put a spotlight on the inheritance bug.

reversefold, 2020-01-11:
Looks like that's the issue, yes. Thanks, I'll try beta2.

dhalbert, 2020-01-15:
Should be fixed by #2504, which will be in the next beta. Please reopen if not.

closed 2020-01-15 by dhalbert