adafruit/circuitpython #2509: "Cannot remount '/' when USB is active." even when USB disconnected (nRF52840)

urish, 2020-01-13:
I'm getting a "Can't remount '/' when USB is active" error, when calling `storage.remount("/", False)` , despite the fact USB is not connected while the command is running.

## Steps to reproduce

The board has to be battery powered, so that it keeps running the code even when USB is disconnected.

1. Get into CircuitPython REPL (over USB-CDC)
2. Type the following code:
```python
import storage
import time
try:
  time.sleep(15)
  success = False
  storage.remount("/", True)
  success = True
except Exception as exc:
  error = exc
```
3. Quickly disconnect the board from USB, and wait for 15 seconds (or more)
4. Reconnect the board and reopen the REPL, and then inspect the value of `success` and `error`. You will see this output:
```
>>> success
False
>>> error
RuntimeError("Cannot remount '/' when USB is active.",)
>>> 
```

frank1119, 2020-01-13:
I think `storage.remount("/", False)` should go in boot.py, because that is being run before code.py and (assumed by me) before the REPL activates

urish, 2020-01-14:
@frank1119 yes, that works. What I'm trying to achieve is to write to the storage without having to tell the user to restart the device with one of the buttons pressed down. It makes sense that we'd be able to remount the storage as writable when the USB is not connected.

ladyada, 2020-01-14:
ah i think that is not possible at this time - we'd need https://github.com/adafruit/circuitpython/issues/544 first

urish, 2020-01-14:
@ladyada if we get someone from our team to look into #544, would that help to move things forward?

ladyada, 2020-01-14:
im sure it would be greatly appreciated! when dan & scott are awake they can comment further :)

frank1119, 2020-01-14:
Maybe setting a flag in nvm (if possible) to test during boot?

urish, 2020-01-14:
@frank1119 that's a good workaround. I tested and the nvm size on the nRF52840 seems to be 8kb. Our use case is uploading new code/assets through Bluetooth, so this will restrict us to around 8kb at a time, but it's still better than nothing. Thanks for the suggestion!

frank1119, 2020-01-14:
@urish I was not suggesting to use nvm as storage, but it is surely a way to think out of the box :-)

urish, 2020-01-14:
@frank1119 yeah, temporary storage :-)

tannewt, 2020-01-14:
@urish I think this issue just needs [`usb_enabled`](https://github.com/adafruit/circuitpython/blob/master/supervisor/shared/usb/usb.c#L56) to be changed. Basing it on a variable set in mount/umount lower in that file would be better. The terms there mean plugged and unplugged and are not mass storage specific.

That being said, I want to add a circuitpython service for uploading and downloading files. The beginnings of it are here: https://github.com/adafruit/circuitpython/blob/master/supervisor/shared/bluetooth.c

I want to do a rev on the API but that is my long term way of doing it.

urish, 2020-01-14:
@tannewt Thanks for the feedback!

The file transfer services look interesting, have you considered how it would play together with the USB storage?

tannewt, 2020-01-14:
Only a bit. My thinking would be that first one wins. So if you are on USB first then BLE would be read-only. If you are on BLE first then USB would be read-only. If you unplug from USB then you can be granted write access too.