adafruit/circuitpython #2514: microcontroller.cpu.temperature doesn't work on Teensy 4.0

rcolistete, 2020-01-14:
"Adafruit CircuitPython 5.0.0-beta.3-6-g926375d99-dirty on 2020-01-10; Teensy 4.0 with IMXRT1062DVJ6A" halts after :
```
import microcontroller
microcontroller.cpu.temperature
```
Thanks for making CircuitPython available on Teensy 4.0 !

manitou48, 2020-01-16:
It seems hardware reset occurs with **calibrationData = OCOTP->ANA1;**  in 
sdk/devices/MIMXRT1062/drivers/fsl_tempmon.c
There is working TEMPMON example in SDK and in Teensy4 core

manitou48, 2020-01-16:
OK this might be the fix:  in sdk/devices/MIMXRT1062/drivers/fsl_tempmon.c before
calibrationData = OCOTP->ANA1;
add  **CLOCK_EnableClock(kCLOCK_Ocotp);**

>>> microcontroller.cpu.temperature
48.1304

ladyada, 2020-01-16:
@manitou48 nice find! would you like to submit a PR?

arturo182, 2020-01-16:
How strange, I definitely tested that code before and it did work, wonder if I removed the clock enabling code somewhere, anyway, @manitou48 if you want to fix it, it should probably be here: https://github.com/adafruit/circuitpython/blob/master/ports/mimxrt10xx/common-hal/microcontroller/Processor.c#L36

manitou48, 2020-01-16:
others can submit a PR ...

PR #2926 (2020-05-18): mimxrt10xx: Fix cpu.temperature crash
Fixes #2514

The TEMP peripheral requires OCOTP to get the calibration data.
commit 058a7312 (2020-05-18): mimxrt10xx: Fix cpu.temperature crash

Fixes #2514
closed 2020-05-18 by tannewt