adafruit/circuitpython #2516: ampy doesn't work with Teensy 4.0

rcolistete, 2020-01-14:
With "Adafruit CircuitPython 5.0.0-beta.3-6-g926375d99-dirty on 2020-01-10; Teensy 4.0 with IMXRT1062DVJ6A", ampy utility doesn't work with Teensy 4.0 :
```
$ ampy -p /dev/ttyACM0 ls
Traceback (most recent call last):
  File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 537, in write
    n = os.write(self.fd, d)
OSError: [Errno 5] Input/output error

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/roberto/.local/bin/ampy", line 10, in <module>
    sys.exit(cli())
  File "/usr/lib/python3.8/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3.8/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3.8/site-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3.8/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3.8/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/roberto/.local/lib/python3.8/site-packages/ampy/cli.py", line 194, in ls
    for f in board_files.ls(directory, long_format=long_format, recursive=recursive):
  File "/home/roberto/.local/lib/python3.8/site-packages/ampy/files.py", line 164, in ls
    out = self._pyboard.exec_(textwrap.dedent(command))
  File "/home/roberto/.local/lib/python3.8/site-packages/ampy/pyboard.py", line 265, in exec_
    ret, ret_err = self.exec_raw(command)
  File "/home/roberto/.local/lib/python3.8/site-packages/ampy/pyboard.py", line 256, in exec_raw
    self.exec_raw_no_follow(command);
  File "/home/roberto/.local/lib/python3.8/site-packages/ampy/pyboard.py", line 248, in exec_raw_no_follow
    self.serial.write(b'\x04')
  File "/usr/lib/python3.8/site-packages/serial/serialposix.py", line 571, in write
    raise SerialException('write failed: {}'.format(e))
serial.serialutil.SerialException: write failed: [Errno 5] Input/output error
```

ladyada, 2020-01-14:
this is a feature, not a bug? 😀we dont want folks using ampy, its for boards without native usb

arturo182, 2020-01-14:
ampy uses raw REPL mode so very likely related to #2515

rcolistete, 2020-01-14:
> this is a feature, not a bug? we dont want folks using ampy, its for boards without native usb

ampy works with Pyboard v1.1/Lite/D, which have native USB (internal file system as USB mass storage).

ladyada, 2020-01-14:
kk, btw we dont automatically permit writing to fs from he interpretter, youd have to remount it

rcolistete, 2020-01-14:
> kk, btw we dont automatically permit writing to fs from he interpretter, youd have to remount it

Ok. So "Mu Editor" seems the recommended IDE, right ?

ladyada, 2020-01-14:
i like it!

dhalbert, 2020-01-15:
Closing as "solved". Please reopen if there's something further to address.

closed 2020-01-15 by dhalbert