adafruit/circuitpython #2517: BLE HID paired connections not working on some Windows 10 peripherals

dhalbert, 2020-01-14:
Copied from #1050.

@urish: 
>Windows 10 does see the device (even though not always - I had to remove the scan_response part from the advertising so that it show it on the list of pairable devices), but disconnects about one second after pairing with a "Driver error" message:

>[image]

>Android works well

@mscosti:
>@urish I'm seeing the same thing as you. At first I thought this was related to changes I have made in an attempt to implement a different HID device incorrectly / missing something (Microsoft Radial Controller / Surface Dial) , But after seeing your report I loaded up this BLE HID Keyboard example verbatim and am seeing the same behavior on Windows 10. Same thing, where it takes a second or two for it to settle with "Driver error".

>[image]

>Android Recognizes the keyboard and am able to type on it as expected

dhalbert, 2020-01-14:
@dhalbert:
>I have been working on BLE bonding over the past couple of weeks (see #2510), and just tested on Windows 10 with a BLE-capable dongle, and did not have this problem. But it also worked before without bonding. So it may be particular to the Bluetooth hardware on your Windows machine(s).@mscosti and @urish, what is your hardware?

@mscosti:
>@dhalbert I'm using a MS Surfacebook 2, which has Bluetooth built in (no dongle)

@dhalbert:
>@mscosti hmm, some kind of pairing issue mentioned here: https://social.technet.microsoft.com/Forums/en-US/31e39461-0b8b-4daa-86d0-010f0bd2feed/microsoft-surface-and-ble-issue?forum=win10itprohardware. The OP said it works with a dongle, but not with the builtin Bluetooth hardware. I assume you're up to date on drivers and the latest version of Windows 10, but if not, that might be worth trying.

>I did not find many similar reports, but very few people are trying to use BLE HID with Windows. It was only fairly recently it started to be better supported.

dhalbert, 2020-01-14:
@mscosti:

> The issue mentioned here sounds very similiar to the one I'm experiencing. https://answers.microsoft.com/en-us/windows/forum/all/problem-with-bluetooth-low-energy-gatt-compliant/ec845657-d885-47e8-a796-cd3ff83a7698
> 
> Unfortunately, the suggested solution that others confirmed working for them (uninstalling Microsoft Bluetooth LE Enumerator and restarting) did not seem to work for me. @urish , I am curious if this works for you, though.
> 
> I'm not sure if its relevant/true/meaningful, but when I look at the device in device manager / HID -> Bluetooth Low Energy Gatt compliant HID device, it says error code 10 in status like in the reported issue above, and the first response says
> 
> According to the error code, the cause of the issue: the device's hardware key contains a "FailReasonString" value, and the value string is displays an error message defined by the hardware manufacturer. If the hardware key does not contain a “FailReasonString” value the message above is displayed.
> 
> As far as I can tell i'm up to date on my drivers. I have 2 other Bluetooth HID devices (Pen digitizer, granted its a MS device), and a PS4 game pad, both of which are able to pair and function just fine.

@dhalbert:

> According to the error code, the cause of the issue: the device's hardware key contains a "FailReasonString" value, and the value string is displays an error message defined by the hardware manufacturer. If the hardware key does not contain a “FailReasonString” value the message above is displayed.
> 
> I think that is just saying that it's failing and we didn't supply a reason string ??
> 
> As far as I can tell i'm up to date on my drivers. I have 2 other Bluetooth HID devices (Pen digitizer, granted its a MS device), and a PS4 game pad, both of which are able to pair and function just fine.
> 
> Are those other devices BLE or classic Bluetooth (it might be hard to tell from the specs)?

dhalbert, 2020-01-14:
@mscosti:

> hmm. Not sure about the PS4 game pad, I think that's older, but the Surface Pen is listed as using Bluetooth 4.0, and in device manager shows up in HID section as Bluetooth Low Energy GATT compliant HID. So, still not 100% sure, but most likely?
> 
> Is there perhaps some arduino BLE HID code I could try on the CPB instead to see if I get the same issue?

@dhalbert:
>How about https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/hid_keyboard/hid_keyboard.ino ? You'll need to install the Adafruit nRF52 BSP if you haven't already.
>[image]

dhalbert, 2020-01-14:
@mscosti asked out-of-band which Bluetooth/BLE dongle is working for me. I am using this one: https://www.adafruit.com/product/1327.

mscosti, 2020-01-16:
Managed to get another couple datapoints today.

- Tried the same circuit python BLE keyboard code on a co workers windows machine (different manufacturer, different Bluetooth chipset even) with _built in_ bluetooth and got the same error (driver error)
- Loaded up the Arduino BLE keyboard demo and was able to successfully connect and have windows recognize and treat it as a keyboard.

[image]

This rules out my laptop not having BLE support or out of date drivers.

It seems there must be something different with how Arduino and Circuit python BLE libraries are advertising? (my terminology might be wrong, I'm not very familiar with the Bluetooth/BLE protocol). Or, at least different enough where Windows is being finicky in this particular scenario.

@dhalbert  Do you have access to a windows 10 machine with BLE built in and so you don't need to use a dongle? I'm getting the feeling that that might be a consistent way to reproduce the issue.

mscosti, 2020-01-16:
Did a little bit of digging. Used the nRF connect android app to scan so I could log what if any differences there was in advertisement data between the advertisement sent while using [this arduino sketch](https://github.com/adafruit/Adafruit_nRF52_Arduino/blob/master/libraries/Bluefruit52Lib/examples/Peripheral/hid_keyboard/hid_keyboard.ino) vs using [this circuitpython example](https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/668b0bcb280a2cfd4c1a5e5ebd90c0c81d78ce52/examples/ble_hid_periph.py) . I kept the device name the same between the 2 so the 2 could be more easily comparable.

Here are some screenshots of what nRF Connect was seeing.
### Arduino Sketch Advertisement (parsed)
[image]

### Arduino Sketch Advertisement (raw data)
`0x020106020A040319C103030312180C094D79204B6579626F617264`
[image]

### Circuit Python Code Advertisement (parsed)
[image]

### Circuit Python Code Advertisement (raw data)
`0x030212180319C1030C094D79204B6579626F617264`
[image]

Looking at the Raw data of each, It seems like the only difference between the two is that the python example code is not sending type `0x01` (flags), and `0x0A` (TX power levels), and associated values for those types. 

It looks like we _are_ setting the default flags in the Advertisement class [here](https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/master/adafruit_ble/advertising/__init__.py#L123) to be General Discoverability and LE Only (brEdrNotSupported) , and I confirmed the field is set to those two on the `ProvideServicesAdvertisement` Object, but it appears that for some reason it isn't making its way out in the actual Advertisement payload? 

My guess is that MS is pickier than most and since it isn't getting those flags it incorrectly makes assumptions about what kind of device it is.

dhalbert, 2020-01-16:
This is really odd because I did the same thing as you, and am seeing the flags field (`0x01` with value `0x06`) in the advertisement. Below is my screenshot. I am using the code below as a test: it's a slightly modified version of `examples/ble_hid_periph.py`. I'm using https://github.com/adafruit/Adafruit_CircuitPython_BLE/releases/tag/3.1.2 (not 4.0.0).  I'm using a very recent build off master of CircuitPython. Do these versions differ from what you are using?

[ble_hid_periph.py.txt](https://github.com/adafruit/circuitpython/files/4069417/ble_hid_periph.py.txt) (renamed to .txt to be able to upload to github)

[image]

dhalbert, 2020-01-16:
I am seeing the "Legacy" problem with no flags with 4.0.0. Will investigate further.

mscosti, 2020-01-17:
CIrcuit Python firmware version: 5.0.0 beta 3: https://github.com/adafruit/circuitpython/releases/tag/5.0.0-beta.3

Not 100% sure what version of Circuitpython_BLE i'm on; whatever is the version that came precompiled in this bundle (and used the bundle for 5.x.x): https://github.com/adafruit/Adafruit_CircuitPython_Bundle/releases/tag/20200111

Looking at the release of CIrcuitpython_BLE though, I would assume it would have to be https://github.com/adafruit/Adafruit_CircuitPython_BLE/releases/tag/3.1.2 since that was the most recent release when the bundle was release? 

So I think we might have been on the same versions?