adafruit/circuitpython #2527: Add CANbus API

tannewt, 2020-01-16:
Add CAN bus API. Very common in cars and robots. https://en.wikipedia.org/wiki/CAN_bus

skpang, 2020-01-16:
Please add CAN and CAN FD support. 

Teensy 4.0 have two CAN 2.0B and one CAN FD port.

jepler, 2020-08-27:
I am working on a CAN implementation for SAM E54 mcus.  It will be called `_canio` and we're working on defining the initial scope and API structure.  There are several different sketches of how the API might look, including https://gist.github.com/tannewt/89ee082de9f5a98675df901cdcfbe951 and http://zaphod.unpythonic.net/canio-provisional-docs/html/shared-bindings/_canio/index.html

CarlFK, 2020-08-27:
I'm new to canbus, so I am treading lightly.  

I would like this to be exposed:
```
32.9.3 CAN mailbox registers
...
Bits 31:16TIME[15:0]: Message time stamp 
This field contains the 16-bit timer value captured at the SOF transmission.
```
p1108
https://www.st.com/resource/en/reference_manual/dm00031020-stm32f405-415-stm32f407-417-stm32f427-437-and-stm32f429-439-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf

I don't know if this is a general canbus thing that is expected to be available across all implementations, or STM32 specific.  

If this is easy to accommodate in the CP module and just happens, great.   

If it looks like this is a problem and thus passed over, can we plan for a way to make it possible for me to subclass so I don't have to maintain a fork of the whole module.  This sounds like a good idea anyway.

I'll gladly help test and put some effort into this.

skieast, 2020-09-01:
Hardware  mailboxes although relatively common arent available on all implementations.  I'm guessing they could be emulated using filters and buffers. I'm definitely not an expert.   Here is a short description on why they are nice,
https://forum.macchina.cc/t/canbus-mailboxes/710
They also have a lot of nice hardware and software for vehicle OBD2 / CAN exploration

closed 2020-11-09 by jepler