adafruit/circuitpython #2531: circuitpython for the steam controller

2bndy5, 2020-01-17:
### Would it be feasible/possible to port CircuitPython to the steam controller's main processor (the [LPC11U37F](https://www.nxp.com/docs/en/data-sheet/LPC11U3X.pdf))?
The real advantage is a wireless remote control that can be programmed to use the [nRF51822](https://www.nordicsemi.com/eng/Products/Bluetooth-low-energy/nRF51822) ([via USART+CTS+RTS and SWDIO/SWDCLK](https://github.com/greggersaurus/OpenSteamController/blob/master/ReverseEngineering/Luna_maiboard_V000456-00_rev3.md#nrf51822)) to communicate with other Bluetooth modules (or hopefully Nordic proprietary protocols like Enhanced Shockburst used by the nRF24L01). In addition to the obvious monotone speakers/buzzers (may be doubling for haptic feedback), analog joystick/triggers, and various buttons, there's an [Invensense MPU6500](https://www.invensense.com/products/motion-tracking/6-axis/mpu-6500/) accelerometer/gyroscope (via I2C), one white LED behind the steam button, and 2 circular trackpads ([TM040040](http://www.cirque.com/glidepoint-circle-trackpads) via 1 SPI bus). All this hardware info and much more can be obtained from [the OpenSteamController project](https://github.com/greggersaurus/OpenSteamController). I know [Valve stopped making them](https://www.theverge.com/good-deals/2019/11/26/20984123/valve-steam-controller-discontinued-sale-price), but they're so easily hackable...

dglaude, 2020-01-18:
I would love that, beep me for testing if ever it exist and I don't notice it... but this seems unlikely.

ladyada, 2020-01-18:
128K Flash / 12KB SRAM makes it not a great target - our current 'minimal' configuration is samd21 with 256K / 32K and its a tight fit :) 

only option that makes sense to us is you could program that chip to send BLE data that could be picked up by one of our BLE circuitpy boards.

2bndy5, 2020-01-18:
Thank you. @ladyada your experience is invaluable. The memory issue was my main concern. I guess I'll try Arduino IDE support instead. The mentioned OpenSteamController project has not started to reprogram the nrf51822 (via SWDIO/SWDCLK) yet, but they have written c++  firmware that can read most of the inputs (except the mpu6500) from the LPC11U37F. 

For those that landed here from Google: this idea is dead, please follow the [OpenSteamController project](https://github.com/greggersaurus/OpenSteamController)

closed 2020-01-18 by 2bndy5

ladyada, 2020-01-18:
yep agreed, there's arduino drivers for all those chips - so ya just need to find a way to program the chip (DFU? SWD?) and find someone with an LPC11 BSP. good luck, and let us know if you find anything, we'll blog it up!

2bndy5, 2020-01-18:
Afraid of going a little off topic here, but I think they're using DFU as the process to reprogram the LPC11U37F is strikingly similar to updating CircuitPython (hold right trigger when plugging in USB and a small HDD is enumerated that contains `firmware.bin`). The one concern that's been lurking in my mind is no hard reset button, but valve most likely left usual testing pads exposed, so maybe there's a solution that involves soldiering.