adafruit/circuitpython #2535: Crash into the HardFault_Handler

Antavatech, 2020-01-20:
Hi, received my new feather M4 express and trying to run one of the sample codes from your website.  I keep receiving the error below:

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

I have tried reapplying the circuit python bootloader "adafruit-circuitpython-feather_m4_express-en_US-4.1.2.uf2" but still receive the same error.  The only lib copied over is neopixel.py, thanks.

[boot_out.txt](https://github.com/adafruit/circuitpython/files/4088328/boot_out.txt)

[code.txt](https://github.com/adafruit/circuitpython/files/4088332/code.txt)

ladyada, 2020-01-21:
huh can you try circuitpy 5.x (the latest release) https://circuitpython.org/board/feather_m4_express/

Antavatech, 2020-01-21:
Hey, I tried that today and it's working.  Also, I ordered 2 FM4Ex and tried the second one today.  Same error message when using CircuitPython 4.1.2, but is now working with 5.0.0-beta.3.

ladyada, 2020-01-21:
ok we fixed whatever bug you had :)
as we're moving toward 5x release soon, we will close this ticket. please keep to 5x!

closed 2020-01-21 by ladyada