adafruit/circuitpython #2537: iMX USB is having caching issues

tannewt, 2020-01-21:
USB in #2532 is having issues. Like not producing the correct data for CDC and MSC. 

https://forum.pjrc.com/threads/59040-CircuitPython-on-Teensy-4!?p=226968#post226968

Will disable the DCache for now and we can fix this and re-enable the DCache after.

arturo182, 2020-01-21:
I haven't had that issue on the Feathers or evks so I think it might be only on your branch :D

hathach, 2020-01-21:
> USB in #2532 is having issues. Like not producing the correct data for CDC and MSC.
> 
> https://forum.pjrc.com/threads/59040-CircuitPython-on-Teensy-4!?p=226968#post226968
> 
> Will disable the DCache for now and we can fix this and re-enable the DCache after.

@tannewt can you sum up the discussion in pjrc forum, seem like they have a strict IP restriction. I tried opera but still cannot pass their firewall.
[image]

arturo182, 2020-01-21:
Ouch, maybe Paul could whitelist your IP.

In the meanwhile, [here's a screenshot](https://user-images.githubusercontent.com/249082/72830000-19874f80-3c80-11ea-8674-4e87b993f7dd.png)

jerryneedell, 2020-01-21:
@hathach in case it is useful to know, I found that I was getting read errors when reading a file from Flash at 2k bytes into the file -- see note in https://github.com/adafruit/circuitpython/pull/2532

tannewt, 2020-01-21:
Here is the specific post:

[image]

hathach, 2020-01-22:
thanks @tannewt @arturo182 @jerryneedell I will try to see if I could help on this issue :) . Another question, where should I find those mentioned scripts **raytrace.py** and **llutm.py** .

@arturo182 It is hard to whitelist since my IP is dynamic :)

tannewt, 2020-01-22:
@manitou48 would you mind posting your scripts you used in testing here?

manitou48, 2020-01-22:
[pystuff.zip](https://github.com/adafruit/circuitpython/files/4099846/pystuff.zip)
I've attached a zip with the circuitpython test scripts.  The scripts ran with with previous release of Teensy4 circuitpython and they run with python3.  here are some performance numbers
`           sinperf   fact   llutm  toggle   raytrace(32x32)
pyboard     13679    2129    1593    13.9 us   5.384 s   micropython v1.5
STM32F405   13114    1885    1984              5.123 s   micropython v1.11
STM32F405   59997    9995    5005    38.4     24.636 s   circuitpython
M4          21973    4028    1984    21.4     10.785 s
T4          31006    3662    5371    43.6     13.178 s
`
FWIW, NXP SDK and Teensy 4 core both copy all of FLASH to RAM as part of startup.  Most NXP examples disable cache.  Teensy 4 has functions to clear sections of cache for DMA and IO use. Here is description of memory layout for Teensy 4:
https://www.pjrc.com/store/teensy40.html

tannewt, 2022-03-09:
This was done in #3522

closed 2022-03-09 by tannewt