adafruit/circuitpython #2538: Configure own SPI Flash

bitShakeMX, 2020-01-21:
Hi, I made my own board based on SAMD21, I could burn the U2F support successfully, but once I tried to install circuit python my board not respond, until I press reset twice to get in bootloader mode.

I'm using this SPI Flash https://datasheet.lcsc.com/szlcsc/1912111437_BOYAMICRO-BY25D40ASOIG_C382741.pdf

and this #define:

// Datasheet: https://datasheet.lcsc.com/szlcsc/1912111437_BOYAMICRO-BY25D40ASOIG_C382741.pdf
#define BY25D40 {\
    .total_size = (1 << 22), /* 4 MiB */ \
    .start_up_time_us = 300, \
    .manufacturer_id = 0x68, \
    .memory_type = 0x40, \
    .capacity = 0x13, \
    .max_clock_speed_mhz = 108, \
    .quad_enable_bit_mask = 0x00, \
    .has_sector_protection = true, \
    .supports_fast_read = true, \
    .supports_qspi = false, \
    .supports_qspi_writes = false, \
    .write_status_register_split = false, \
    .single_status_byte = false, \
}

I got out the JEDEC data with the SPI Flash library, I'm tried to format the memory with the example (SdFat_format) over arduino, but the program stuck.

Maybe the SPI Flash is low quality or maybe I'm missing something?

help!

tannewt, 2020-01-21:
I'd recommend using a logic analyzer to sniff the data lines to the flash. That way you can easily understand how far the process is getting. This clip can make it easier: https://www.digikey.com/product-detail/en/pomona-electronics/5250/501-1311-ND/745102

The other way to debug it is to see how far it is getting from the code side. However, it is much trickier.

I would expect it to work but it might have a quirk of some sort that the config isn't accounting for.

dhalbert, 2020-01-23:
@bitShakeMX Feel free to reopen this if you need further help.

closed 2020-01-23 by dhalbert

bitShakeMX, 2020-01-24:
Thanks for the reply! I will test with logic analyzer