adafruit/circuitpython #2540: Compilation of micro python failing on Mac OSX

tsupplis, 2020-01-21:
Build from master branch.

Building micro python (make in ports/unix) fails with:
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
```
In file included from ../../py/emitnx64.c:13:
../../py/emitnative.c:467:32: error: pragma diagnostic pop could not pop, no matching push [-Werror,-Wunknown-pragmas]
        #pragma GCC diagnostic pop
```
Suggested fix on line 462 of 

```
        #pragma GCC diagnostic push
        #pragma GCC diagnostic ignored "-Wcast-align"
        mp_emit_glue_assign_native(emit->scope->raw_code,
            emit->do_viper_types ? MP_CODE_NATIVE_VIPER : MP_CODE_NATIVE_PY,
 ```

tannewt, 2020-01-22:
Why are you building CircuitPython for unix? What version of Mac OSX are you on?

tsupplis, 2020-01-22:
Building unix version to 
1. execute tests
2. work on extensions

Below info on os and compiler:

Darwin betel 19.2.0 Darwin Kernel Version 19.2.0: Sat Nov  9 03:47:04 PST 2019; root:xnu-6153.61.1~20/RELEASE_X86_64 x86_64

Apple clang version 11.0.0 (clang-1100.0.33.17)
Target: x86_64-apple-darwin19.2.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

tsupplis, 2020-01-22:
diagnostic push and pop pragmas are not balanced. gcc closes an eye on it. clang does not.

tannewt, 2020-01-22:
Thanks! It must have been added with a compiler update. I've built them before and not had a problem. I am now though so I'll reopen the PR.

tannewt, 2020-01-22:
Fixed by #2541

closed 2020-01-22 by tannewt