adafruit/circuitpython #2544: STM32: implement rotaryio

scs217, 2020-01-21:
`rotaryio` not currently implemented in circuitpython for STM32 based boards (e.g., STM32F405).

scs217, 2020-01-22:
I'll see what I can do in the way of progress on this.

scs217, 2020-01-23:
It looks like I can implement this the way it's done in AMTEL-SAMD or use the TIM_ENCODER functionality provided by the STM32. I saw this is in the comments for SAMD51 to be implemented later on.

Can I have some guidance on which way the good people at adafruit would like this to be implemented?

hierophect, 2020-01-23:
If you are comfortable with the STM32 TIM_ENCODER then I think that would be preferable. I notice that the Atmel implementation has a todo for implementing it in the native hardware - might as well skip that step if we can.

scs217, 2020-01-24:
Either method is completely new to me so I'll try and do it the right way!