adafruit/circuitpython #2547: CircuitPython crashes when saving Prop-Maker Lightsaber code

CudaCoreRoo, 2020-01-22:
CircuitPython crashed and went into safemode when saving the code from this website from Mu: https://learn.adafruit.com/lightsaber-featherwing/software

I've been attempting to get this lightsaber to work for a while with no success, but this is the first time I've had a crash so something must have been changed within the last hour because I just updated all the files in the library folder. I'm running the latest version of CircuitPython 4 and using the latest CircuitPy 4 library files. The libraries being used are the ones listed on the adafruit page linked above.

ladyada, 2020-01-22:
hi are you the same person who posted in the forum? if so paste the link here

CudaCoreRoo, 2020-01-22:
That's correct. I've been experiencing issues and was told on a reddit thread I started that I should try using an older neopixel library file, so I did that, but now it seems circuitpython 4 just completely crashes when saving that lightsaber code. using CircuitPy 3 doesn't crash, but it does present me with the same 'no sound, no light effects' issue I've been having for a week. 

Adafruit thread: https://forums.adafruit.com/viewtopic.php?f=57&t=160960

Here's the reddit thread too: https://www.reddit.com/r/adafruit/comments/es6zwb/cannot_get_adafruit_lightsaber_to_work_right/

dhalbert, 2020-01-23:
We have backed out a change to `neopixel` that we think should fix this problem. Could you try this release: https://github.com/adafruit/Adafruit_CircuitPython_NeoPixel/releases/latest (5.0.0)? It is not in the bundle yet. Thanks!

CudaCoreRoo, 2020-01-24:
While it does fix the problem of circuitpython crashing it does not fix my overall problem of my lightsaber not functioning as it should, though that problem doesn't have anything to do with the crashing so I don't expect any replies on that here.

dhalbert, 2020-01-24:
Thanks, I will close this and hope you can solve the other problems in the forum thread.

closed 2020-01-24 by dhalbert

tannewt, 2020-01-24:
@CudaCoreRoo Noe and I got back to you on the forum so we should be able to get you sorted out there.