adafruit/circuitpython #2549: Modifying gcc file for making CircuitPython

GathererA, 2020-01-24:
Hello! I am attempting to add an option into the gcc call that CircuitPython uses to compile one of the modules. I am working to include a separate shared library for wrapping C++ into C for use in CircuitPython, and this requires citing a .so file that was generated with g++ when calling gcc. Is this possible, and if so, where do the gcc file falls live? I found .p files that are automatically generated when I attempt to build a board, but I’m not sure if these include the gcc call.

dhalbert, 2020-01-24:
The CircuitPython compilation process generates a bunch of `.o` files that are linked together into a single `.elf`. `.so` is for run-time dynamic linking, which I don't think makes sense here. For instance, here's the final linking step from `ports-atmel-samd/Makefile`:
```make
$(BUILD)/firmware.elf: $(OBJ) $(GENERATED_LD_FILE)
	$(STEPECHO) "LINK $@"
	$(Q)$(CC) -o $@ $(LDFLAGS) $(OBJ) -Wl,--start-group $(LIBS) -Wl,--end-group
	$(Q)$(SIZE) $@ | $(PYTHON3) $(TOP)/tools/build_memory_info.py $(GENERATED_LD_FILE)
```

pdetrempe, 2020-01-24:
Looks like there's an area of the makefile where the process imports libraries for linking. Would it be possible to link libraries by appending them to this list?

LIBS := -lgcc -lc

```# Use toolchain libm if we're not using our own.
ifndef INTERNAL_LIBM
LIBS += -lm
endif```

GathererA, 2020-01-27:
To build on what Paul is saying, our MakeFile isn't able to even read .o file, so I'm sure we're just doing something wrong in converting our C++ code into a library that the C can read. Do you have any recommendations on creating a static .o file out of C++ code that can be integrated into CircuitPython?

dhalbert, 2020-01-27:
Can you add some extra rules to the global Makefiles so that you can compile the `.cpp` file(s) with `g++` and produce a .o as is already done with `.c`? You;ll need to provide C-style linkage to the symbols you need in the C++ code.

GathererA, 2020-01-28:
We are trying to modify the ports/atmel-samd MakeFile to include the new object file: 
`LDFLAGS = $(CFLAGS) -nostartfiles -fshort-enums -Wl,-nostdlib -Wl,-T,$(GENERATED_LD_FILE) -Wl,-Map=$@.map -Wl,-cref -Wl,-gc-sections -specs=nano.specs
LIBS := -lgcc -lc -ldetumble
LDFLAGS += -L/Volumes/CircuitPython/circuitpython/ports/atmel-samd`

where the new LDFLAG is the directory of where we have put the the new object file (.o) that we are generating elsewhere, but this is giving us the error:

`/usr/local/Caskroom/gcc-arm-embedded/7-2018-q2-update/gcc-arm-none-eabi-7-2018-q2-update/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: cannot find -ldetumble
collect2: error: ld returned 1 exit status
make: *** [build-kicksat-sprite/firmware.elf] Error 1`

Where should we include the new object file in the MakeFile, and where should the physical .o file live?

dhalbert, 2020-01-28:
To add a `.o` file, you could just add
```make
OBJ += something.o
```

You can find the other additions to `OBJ` in the `Makefile`.

_But_, a C++ object file may need `libstdc++` or other libraries. And, if that `.o` may need to have been compiled with other switches that define how subroutine linkage is done, etc. So supplying a `.o` without knowing how it was built is not necessarily going to succeed. That is why I'm suggesting compiling the C++ code with the same flags, not separately.

See https://github.com/micropython/micropython/issues/3472#issuecomment-350162099 and that entire issue for how MicroPython has sometimes incorporated C++ code.

GathererA, 2020-01-30:
Maybe adding a shared library is the wrong idea. I just compiled our .cpp code with gcc -lstdc++ as a flag. Would it be possible to add -lstdc++ somewhere in the make file, perhaps in LIBS?

dhalbert, 2020-01-30:
Yes, certainly, just add it to `LIBS` with `+=`, and see whether it works.

GathererA, 2020-01-30:
This doesn't seem to work, since we can't use c++ syntax like `#include <cmath>` rather than `#include "math.h"` in our files. Any other recommendations of how to switch to g++, since g++ should be able to compile both c and c++ files, correct?

tannewt, 2020-01-30:
You'll like need to modify the Reset_Handler as well for C++ because we don't do any of the C++ static initialization there.

GathererA, 2020-01-30:
Do I have to modify the MakeFiles in the samd51 build chains? `/circuitpython/ports/atmel-samd/asf4/samd51/gcc/Makefile`

dhalbert, 2020-01-31:
Yes, and it would be nice to add `CXX` definitions and C++ rules to the `py/*.mk` files.

I would suggest that you find an existing mixed C and C++ program to emulate. There are many details: this will not be that easy without previous experinece. I'm wondering about the C++ you're trying to compile. Could you supply a pointer to it?

GathererA, 2020-02-04:
We'd like to be able to call a c++ function from CircuitPython through a C wrapper. Here's a directory with the .cpp file and the .h file that we'd like to integrate into our CircuitPython build. We've also tried adding the files to our MakeFile on the port level through including them on the -INC call, which hasn't worked either. Fundamentally, our code also depends on the use of Eigen, which is a C++ header-only library. Any other ideas for how we could integrate this? We've tried shared libraries, archive files, and now building with the MakeFile, and none have panned out.

CAPS123987, 2025-04-13:
i suppose no further development  :(. Do you still have the modified Makefile?