adafruit/circuitpython #2566: Add TileGrid example that uses the built in font to display text

tannewt, 2020-01-31:
Similar to terminalio but without using it.

caternuson, 2020-01-31:
from discord:
```
Maybe a multi-line Hello World example that does not use Label would make it clearer how terminal style text could work.
```

caternuson, 2020-01-31:
I'm not sure I understand what can/can't be used here? Like how far down the rabbit hole are we going?

Wouldn't a multi-line example without using `Label` essentially just be a recreation of the code in `Label._update_text()` ?

jepler, 2020-01-31:
@caternuson This stems from discussion over the memory usage of Label.  There is a way to display text using a built-in font that uses much less RAM memory than Label does, by having a single TileGrid rather than a TileGrid for each character.

caternuson, 2020-01-31:
@jepler Ah, thanks. I think that was the clarification I was after.

jepler, 2020-01-31:
```import displayio
class GridLabel(displayio.TileGrid):
    def __init__(self, font, *, text=None, max_size=None, color=0xffffff,
                 width=None, height=None, **kwargs):
        if not text and not max_size:
            raise RuntimeError("Please provide a max size, or initial text")
        cwidth, cheight = self.calc_size(text)
        width = width or cwidth
        height = height or cheight

        self._width = width
        self._height = height
        self.font = font
        self.palette = displayio.Palette(2)
        self.palette.make_transparent(0)
        self.palette[1] = color

        tile_width, tile_height = font.get_bounding_box()[:2]
        super().__init__(width=width, height=height, tile_width=tile_width,
                         tile_height=tile_height, bitmap=font.bitmap,
                         pixel_shader=self.palette,
                         default_tile=self.font.get_glyph(ord('Z')).tile_index,
                         **kwargs)
        self._update_text(text or '')

    def calc_size(self, text):
        start = 0
        width = 0
        height = 1
        while True:
            end = text.find('\n', start)
            if end == -1:
                width = max(width, len(text)-start)
                break
            width = max(width, end-start)
            height += 1
            start = end + 1
        print(width, height)
        return (width, height)

    def _update_text(self, text):
        self._text = text
        r = c = 0
        space = self.font.get_glyph(ord(' ')).tile_index
        for codepoint in text:
            if codepoint == '\n':
                for i in range(c, self._width):
                    self[i, r] = space
                c = 0
                r = r + 1
                continue
            if c == self._width:
                c = 0
                r = r + 1
            if r >= self._height: break
            self[c, r] = self.font.get_glyph(ord(codepoint)).tile_index
            c = c + 1
        while r < self._height:
            for i in range(c, self._width):
                self[i, r] = space
            c = 0
            r = r + 1
        
    @property
    def text(self):
        return self._text

    @text.setter
    def text(self, new_text):
        if new_text != self._text:
            self._update_text(new_text)
```

gmeader, 2020-02-01:
I'm the newbie that made the request in Discord. I was looking for a complete working example/demo that would run as code.py and be included in the /examples folder of the libs. I guess this class is the first step towards that.

jepler, 2020-02-01:
Updated version of gridlabel (there was at least one bug, hopefully fixed): https://gist.github.com/e7a159b28048dc8637f69db17caba5ea
A silly little demo that I ran on my pygamer: https://gist.github.com/4e17ebc6b3a6e762856d2338fdfb99d9

@gmeader Being new to something is an exciting time, enjoy it!

jepler, 2020-02-03:
@gmeader We discussed this in the CIrcuitPython chat, and @tannewt would like to see this made into a pull request in the https://github.com/adafruit/Adafruit_CircuitPython_Display_Text -- both this module and an example.

Would you be interested in helping us out on this?

gmeader, 2020-02-03:
Yes, happy to help.

gmeader, 2020-02-04:
I think GridLabel should have a "color" property (not just the constructor param) that may be set and read. 
Perhaps even a background color property that also may be set to transparent as well?

jepler, 2020-02-04:
Thank you!

I definitely agree with adding a "color" property.  This would make the GridLabel more like the existing Label, which has a color property.

I think adding a background constructor and property is fine, if it is useful.

kmatch98, 2020-05-19:
Here is another example of using text in a tileGrid: https://github.com/kmatch98/simpleTerminal

In modifying a text editor for use with CircuitPython (see the [modules branch of the MicroPython editor pye](https://github.com/robert-hh/Micropython-Editor/tree/modules)), I developed a [`simpleTerminal` class](https://github.com/kmatch98/simpleTerminal) that creates a tileGrid of text with some additions for cursor control.  It has a wrapper class `editorTerminal` that handles some VT100 style commands.  

--------
Performance-related issue for `label`:

As mentioned by @jepler and @caternuson, the performance of the `tileGrid` is much superior to using `label`.  I think this stems from the fact that `label` handles proportional fonts while `tileGrid` can only handle fixed width fonts.  

I'm struggling with the performance of `label`.  It seems like there should be a better way, rather than combining so many tileGrids each holding just one character.  Anyone have any suggestions on a path forward?

tannewt, 2020-07-14:
Is this done?

caternuson, 2020-07-14:
I think so? Looks like it became a PR?

skerr92, 2022-10-16:
This appears to have been fixed/done back in 2020. see [Display_text #75](https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/issues/75) PR

closed 2022-10-16 by caternuson