adafruit/circuitpython #2567: Leaving nRF52840 plugged into Mac when it sleeps crashes the Mac

tannewt, 2020-01-31:
This was fixed on SAMD (I thought) in shared code. It relies on a flush that may make it fail on nRF.

https://github.com/adafruit/circuitpython/commit/561fdfb279c45f7a9ff69a2b2e71ac272cb6aaef#diff-880d63a1c107dd299a7ed80251505a9eR235

tannewt, 2020-01-31:
Note, it's the Mac that crashes not the nRF52840.

dhalbert, 2020-01-31:
The power of microcontrollers.

iayanpahwa, 2020-02-03:
Interesting, how exactly it's happening? I'm curious and willing to reproduce .

tannewt, 2020-02-03:
@iayanpahwa Plug in a nRF52840 running CircuitPython so that it shows up as CIRCUITPY and then sleep the computer, wait 5-10 minutes, open the computer and it should work. Sleep it again and wait another 5-10 minutes and try to wake the computer. It should start up and say that it crashed waiting for a SCSI time out.

iayanpahwa, 2020-02-04:
@tannewt thanks for sharing, I'm thinking what's the exact reason for this crash?

tannewt, 2020-02-04:
The mac is trying to repeatedly start the external mass storage device back up and then timing out. It shouldn't crash like it does.

tannewt, 2020-02-12:
This is also happening with atmel-samd boards. I suspect something with TinyUSB broke this.

iayanpahwa, 2020-02-12:
@tannewt if that comes out to be the case, do you think using an internal fork of TinyUSB makes more sense?

tannewt, 2020-02-12:
@iayanpahwa I'll just fix TinyUSB. :-) I plan on looking at it later while I watch Ask An Engineer.

tannewt, 2020-02-14:
I tried to reproduce this and only managed to do it once. Generally it looks ok. @hathach also tried to reproduce and was unable to.

I'll leave this open in case someone else sees it but move it to the Long Term milestone.

tannewt, 2020-02-22:
This is still happening to me with nrf52840 boards and it's driving me bonkers. I'll look more into this. It may have something to do with the SD being on and periodically advertising. The zip is my full CIRCUITPY contents on a Feather nRF52840 with a battery plugged in along with usb attached.

[20200222-simple-temp.zip](https://github.com/adafruit/circuitpython/files/4239690/20200222-simple-temp.zip)

tannewt, 2020-02-22:
Looks like an old-ish build: `Adafruit CircuitPython 5.0.0-beta.4-111-g89829aea8 on 2020-01-30; Adafruit Feather nRF52840 Express with nRF52840`

tannewt, 2020-02-23:
Ok, I've replicated this twice with: `Adafruit CircuitPython 5.0.0-beta.5-144-gbd9572c8a on 2020-02-22; Adafruit Feather nRF52840 Express with nRF52840`

@hathach The missing piece was to eject the drive before sleeping the computer but leave the CIRCUITPY device connected. Wait a few minutes until the Mac stops talking to the device and then wake the Mac up. It'll keep requesting something that will eventually timeout and restart the computer. I have Beagle traces from each time I can share to those interested.

PR #2653 (2020-02-25): Fix Mac crash when waking up with an ejected CIRCUITPY
We now correctly set the reason for the unit not being ready and
always start the unit.

Fixes #2567
commit 876c646d (2020-02-25): Fix Mac crash when waking up with an ejected CIRCUITPY

We now correctly set the reason for the unit not being ready and
always start the unit.

Fixes #2567
closed 2020-02-25 by tannewt