adafruit/circuitpython #2573: Date limit for time.localtime() function

CedarGroveStudios, 2020-02-02:
The 32-bit signed integer value used when converting seconds-since-epoch to structured time using `time.localtime()` causes a date limit of 2038-01-19 03:14:07 before overflowing.
```python
Adafruit CircuitPython 5.0.0-beta.4 on 2020-01-22; Adafruit Pybadge with samd51j19
>>> 
>>> # test an integer value of 2^^31 (maximum 32-bit signed integer +1)
>>> mk = int(math.pow(2, 31))
>>> time.localtime(mk)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OverflowError: overflow converting long int to machine word

>>> # test an integer value of 2^^31 -1 (maximum 32-bit signed integer value)
>>> mk = int(math.pow(2, 31)) - 1
>>> time.localtime(mk)
struct_time(tm_year=2038, tm_mon=1, tm_mday=19, tm_hour=3, tm_min=14, tm_sec=7, tm_wday=1, tm_yday=19, tm_isdst=-1)
>>>
```
Won't be an issue for current time values for a while. It just limits predictive date math calculations.

dhalbert, 2020-02-06:
Y2038 is the next Y2K.

CedarGroveStudios, 2020-02-06:
Indeed. Suggest that the limitation is noted somewhere in the time module documentation. Python 3.6.3 for Windows overflows sometime after year 2514 (35-bit integer?), but their documentation lacks specificity.

closed 2026-11-13 by dhalbert