adafruit/circuitpython #2578: Gameduino 3X native module

jamesbowman, 2020-02-03:
Gameduino 3/3X are BridgeTek 810/815 emebdded GPUs with touchscreens.

Support for them can be done entirely in Python. Class diagram:

[image]

By replacing one module (eveL) with a native module, the performance-critical methods are 4-7X faster, measured on a Metro M4.

Python code:
https://github.com/jamesbowman/py-bteve

C module for eveL:
https://github.com/jamesbowman/py-bteve/tree/master/circuitpython-shared-bindings-eveL

ladyada, 2020-02-03:
nice! want to do a PR?

jamesbowman, 2020-02-03:
Yes!

ladyada, 2020-02-03:
yay! feel free to start a draft PR if you'd like - we would only have it turned on for our larger boards like the M4 and nrf52840

jamesbowman, 2020-02-04:
Done in PR #2581 

Enabled only for metro_m4_express and metro_nrf52840_express.

dhalbert, 2020-02-13:
Fixed by #2581.

closed 2020-02-13 by dhalbert