adafruit/circuitpython #2580: Two nRF boards cannot be plugged in concurrently on Mac

tannewt, 2020-02-04:
Their drives both have the same UUID which means they don't show up next to one another.

dhalbert, 2020-02-04:
I'm not seeing this on MacOS with two random CPB's. `diskutil info -all` shows different UUIDs for the two volumes.

On Linux the `SerialNumber` that shows up in `/var/log/syslog` is different for the two boards. If you plug the same boards into Linux, are the `SerialNumber` values the same?

[image]

tannewt, 2020-02-04:
Weird! I'm on 10.15.2 still. One was an itsy bitsy and the other a Clue. Will update tonight. My computer could be in a bad state.

tannewt, 2020-02-06:
I think I was thrown off by only one entry in Finder under locations. If I hit computer I can see two. I don't think the serial numbers are actually the same. I was just confused by how mac was handling it. I swear is used to show two in finder. Oh well.

closed 2020-02-06 by tannewt