adafruit/circuitpython #2585: Phantom typing & exiting to REPL on CLUE and CP BLE

siddacious, 2020-02-04:
My CLUE board is experiencing some phantom input and is going into the REPL when a program is done running without input from me. I suspect the two are related.

Test setup:
MacBook Pro 13" w/ touchbar running **Mojave 10.14.6 (18G87)**
Micro USB cable from adafruit: https://www.adafruit.com/product/4148
CalDigit TS3 Plus Thunderbolt 3 Dock

I was testing Dylan's code by copying over files using `cp -R` from his repo, as well as some libs from the bundle. When it first occurred I hadn't used an editor to write anything to the drive

* Dylan's code exits because I don't have all the libs
* I get the 
`Press any key to enter REPL. Use CTRL-D to reload.`

a bit after it adds
```Auto-reload be on. Put er files on USB to weigh anchor, er' bring'er about t' the REPL t' scuttle```
After 30 seconds it repeats the "Press any key to enter the REPL use CTRL-D to reload.
* without  any input from me it goes into the REPL after approximately 30 seconds.
, and I see the following (transcribed, may be some minor typos)
"
```
Adafruit CircuitPython 4.0.0-beta.4-1-27-g67440acd3 on 2020-02-03; Adafruit CLUE nRF52840 Express with nRF52840
>>>
raw REPL; CTRL-B to exit
>
```
 I've tried a few more times and sometimes the timing or order is different, but the end result is always that I end up in the raw REPL. The big issue is that writing to the drive by saving an edit to code.py does not reload

I believe I reproduced it on a CP BLE though it's less certain because it doesn't have a screen to send REPL output to. The behavior I saw was that the previously loaded program that was flashing the neopixels stopped.

I left my clue plugged in over night and saw the following in the morning:
[image]

I reset it to test again and saw this which mentions a keyboard interrupt:
[image]

Of course now that I've taken the time to create this ticket I can't reproduce it anymore. I'm not sure what I may have changed between this morning and now but things seem to be working ok for now 🤷‍♂ 

I'll keep trying to reproduce it again

siddacious, 2020-02-04:
The above images will appear right side up when clicked

tannewt, 2020-02-04:
Try reproducing it without the dock too. It feels similar to modem manager even though this is a mac.

ladyada, 2020-02-04:
M105 was a model of Palm... https://en.wikipedia.org/wiki/Palm_m100_series seems unlikely but if ya had the sync manager on that would probably send something like "M105" over and over :D

siddacious, 2020-02-05:
I _think_ this is being caused by Cura:
https://ultimaker.com/software/ultimaker-cura

I'll double check tomorrow, but my CLUE was behaving  for most of the day, and shortly after opening Cura it started having the issue I reported again. 

It wouldn't surprise me if Cura is doing some sort of scanning for serial ports that look like printers.

ladyada, 2020-02-05:
https://community.ultimaker.com/topic/14915-print-stops-on-passing-m105-serial-command/

FoamyGuy, 2020-02-05:
Not with a CLUE, but I have also witnessed Cura cause interruption and other weirdness with connected Circuit Python devices.

ladyada, 2020-02-05:
technically - its anything based on Marlin https://marlinfw.org/docs/gcode/M105.html

AndrewTribble, 2020-02-05:
Sat down with a serial sniffer for a while and found Cura is doing a bit of hyper-specific handshaking when it connects to a device.  One of the packets it sends is "IRP_MJ_DEVICECONTROL: IOCTL_SERIAL_SET_TIMEOUTS" which contained "00 00 00 00 00 00 00 00 e8 03 00 00 00 00 00 00 80 96 98 00"  when this message is played back from a python script I wrote, we get kicked into repl every time.   

I suspect that "\x03" in the center of the message is the issue.  Cura is sending Ctrl+C as part of the set up procedure, and then from then on we get stuck in REPL.

ladyada, 2020-02-05:
wow good detective work :D i think we just have to warn folks Cura is very 'nosy'

siddacious, 2020-02-06:
wow @AndrewTribble great job! Thanks for tracking this down.

I'm content to close this.

closed 2020-02-06 by ladyada

siddacious, 2020-02-06:
For the sake of documenting the connection, this is issue in the Cura repo seems to be the one that represents the underlying issue:

https://github.com/Ultimaker/Cura/issues/5207

jfurcean, 2020-12-31:
@siddacious thanks for linking to that issue. After taking a closer look, if you disable the "USB Printing" plugin in CURA the issue appears to go away. 

To disable in Cura go to "Marketplace" -> "Installed" tab and uncheck "USB Printing".

kjgoodwin299, 2021-09-02:
Thank you everyone!  I just ran into this exact problem playing around with my new macro keyboard.  I'm new to circuit python and of course chose Cura as the application I would use as a test!  disabled USB printing in marketplace and solved the problem.  So it's not me writing bad code after all!

BenSanborn, 2022-01-20:
Thank you everyone who helped track this down, and the USB Printing fix. I was going nuts trying to find out why my Macropad broke!