adafruit/circuitpython #2587: Add The STM32 Port To Docs Module Support Matrix

sommersoft, 2020-02-05:
The STM32 port isn't being included in the module [Support Matrix on RTD](https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html). 

The port needs to be added here:
https://github.com/adafruit/circuitpython/blob/master/docs/shared_bindings_matrix.py#L29

_Note: could possibly include additional tweaks [here](https://github.com/adafruit/circuitpython/blob/master/docs/shared_bindings_matrix.py#L137) to handle chipset definitions._

I marked it as "Long term", because I'm not sure the milestone that STM is aligned to.

hierophect, 2020-02-05:
Sorry, I haven't done much docs stuff. STM32 is pretty close to NRF in terms of how mpconfigboard.mk is laid out. Is there anything else that I need to change, beyond adding stm32 to this file?

sommersoft, 2020-02-05:
@hierophect no need to be sorry, in my opinion. Its a largely undocumented step, and not easily visible. Could probably be added to the "How to add a new board" learn guide.

Adding the port's folder name to the `SUPPORTED_PORTS` list is the starting point. Then I would build the docs and check that everything is correct. If not, the _Note_ above is probably related to the failure point. I haven't been following the STM code-base that closely, so I'm not immediately aware of any idiosyncrasies.

hierophect, 2020-02-05:
Is there a set of instructions for building the docs locally?

sommersoft, 2020-02-05:
I've always just opened and copied the CI step. 🤣 https://github.com/adafruit/circuitpython/blob/master/.github/workflows/build.yml#L63

Python environment can be tricky. IIRC, I have a pyenv setup for building. Here are the deps:
```
pip install Sphinx sphinx-rtd-theme recommonmark sphinxcontrib-svg2pdfconverter 
```

hierophect, 2020-02-05:
Sounds good, thanks for the advice. I'll set up an environment and see how it goes.

hierophect, 2020-02-06:
@sommersoft my control build (no changes) is not working, unfortunately:
```
Warning, treated as error:
RSVG converter command 'rsvg-convert' cannot be run. Check the rsvg_converter_bin setting
```
Google says this might be related to a missing dependency for sphinxcontrib-svg2pdfconverter. But why would that be any different than the remote build?

hierophect, 2020-02-06:
Nevermind, just had to install librsvg. Didn't realize I had to do that with brew rather than pip.

PR #2598 (2020-04-24): Add stm to docs matrix
As per title. Removes a number of build flags as redundant and reworks the stm port `mpconfigport.mk` to match. Adjusts `shared_bindings_matrix.py` to recognize STM32 family categories. Adds a minor addition to porting.rst to explain how build flags impact porting. 

Builds locally without issue, and correctly lists modules for all STM32 boards in the support matrix list, including the thunderpack (which enables NVM) and the H7 and F7 (which have restricted modules). 

Resolves #2587 
Resolves #2458
closed 2020-04-24 by dhalbert